Ultimaker/Cura #16860: Golf 7

raziel21183, 2023-10-01:
### Cura Version

5.4.0

### Operating System

windows 10

### Printer

Anycubic i 3 mega s

### Reproduction steps

0,2 Qualität
Stützstrucktur Baum

### Actual results

?

### Expected results

?

### Add your .zip and screenshots here ⬇️

[AI3MSPRO_Volkswagen Golf VII GTI 5 Doors 2014 karosserie.zip](https://github.com/Ultimaker/Cura/files/12776804/AI3MSPRO_Volkswagen.Golf.VII.GTI.5.Doors.2014.karosserie.zip)

GregValiant, 2023-10-01:
Thanks for the report.
I can't get it to slice either.  There doesn't appear to be any errors in the model.  
This appears to be a duplicate of the standard slicing bug in 5.4.0.

I'll mark this as a duplicate and the Cura team will take a look.
(It's a long slice, but it sliced fine in 4.13.1 and 5.2.2)

raziel21183, 2023-10-04:
Vielen Dank für die schnelle Antwort

closed 2024-02-26 by raziel21183