Ultimaker/Cura #16861: Slicing issue with the auxiliar tower for 2 colors.

su-luq, 2023-10-01:
### Cura Version

5.4.0

### Operating System

WIndows 10

### Printer

Geeetech A30T

### Name abnormal settings

I've activated the auxiliar tower. Without the auxiliary tower it works.

### Describe model location

No, I have the same error

### Describe your model

It works without the 2 colors auxiliar tower. But I have colors mixing.

### Add your .zip here ⬇️

[054- Caixa RPi3 MMDVM-SuportAntena.zip](https://github.com/Ultimaker/Cura/files/12776956/054-.Caixa.RPi3.MMDVM-SuportAntena.zip)

GregValiant, 2023-10-01:
Thanks for the report and the file.
There is a bug in the Bed Adhesion settings that can interfere with the "Prime Tower Brim".
If you set the "Bed Adhesion" to "Brim" and the "Brim Distance" to 2.0 (needs to be done for each extruder) it will act much like a skirt and should allow the model to slice.

I'll leave this open and the Cura team will take a look.

su-luq, 2023-10-01:
Thanks, it works.

Enviado el: domingo, 1 de octubre de 2023 15:11
Para: Ultimaker/Cura ***@***.***>
CC: Marcel·lí Subirana ***@***.***>; Author ***@***.***>