Ultimaker/Cura #16864: Nothing slices AT ALL in Beta UltiMaker Cura 5.5.0-beta.1+96ccd5

j-snijder, 2023-10-02:
### Cura Version

UltiMaker Cura 5.5.0-beta.1+96ccd5

### Operating System

windows 11

### Printer

ender 3 pro

### Name abnormal settings

nothing

### Describe model location

standard

### Describe your model

part of alack stack model, sliced perfectly in previous versions untill now

### Add your .zip here ⬇️

[CE3PRO_HingeRight.zip](https://github.com/Ultimaker/Cura/files/12780267/CE3PRO_HingeRight.zip)

This file sliced perfectly in previous versions but as it seems at every upgrade of Cura the not slicing bug is getting worse.

Why not stop trying to fix what's not fixable and go back to v4.13 and rebuild from there, check what who did wrong, because this is going to cost you all your users in the long run !!

MariMakes, 2023-10-02:
Hey @j-snijder,

Where did you get a Cura 5.5.0-beta.1+96ccd5 build?
We are not ready to release the Beta yet, I'm not surprised it's not ready to be used yet.

j-snijder, 2023-10-02:
https://github.com/Ultimaker/Cura/actions/runs/6322516836,, anyone can find this I guess ?

MariMakes, 2023-10-02:
So yes, everyone can see the runs, to build Cura releases. 
But they should not be accessible if you are not part of the Cura team. 

If I'm not signed this is what I see when I access the runs. 
[image]

To me, it doesn't seem like a bug in UltiMaker Cura. 
Because you are using a build that is not fit for the public to be used. 
Please return to using Cura 5.4, or wait until Cura 5.5 is released. 

I'll be closing the issue.

closed 2023-10-02 by MariMakes

j-snijder, 2023-10-02:
That's what I see also, although not member of your team :)
And when I click on the exe or msi it downloads and I can install, if it's not possible (or shouldn't be) then how in the name I could get it ?

Jan-Snijder, 2023-12-24:
SOLVED ! damaged ? file in configuration files.