Ultimaker/Cura #16868: Layer modification Start/Stop printing at height

santanig5, 2023-10-02:
### Is your feature request related to a problem?

I am attempting to print a portion of a model for sizing, I need to maintain the print orientation but would only like to print a certain number of layers, I have found no way to accomplish this short of splitting the model. 
[image]

### Describe the solution you'd like

As a long time S3D user there was a feature where one could set the layer height to either start or stop printing.
[image]

### Describe alternatives you've considered

splitting the model in 3D Builder, Banana Split Plugin
[image]

### Affected users and/or printers

Everyone

GregValiant, 2023-10-02:
You gave up too soon.  That first image you posted with the support blocker...
Select just the blocker.
Select the "Per Model" tool on the left toolbar (it's directly above the "Support Block" tool).
Select "Modify Settings for Overlaps"
Select "Cutting Mesh"
Select "Add Settings"
Select: Wall Count, Top Layers, Bottom Layers, and Infill Density.

Here....1/2 of a beer bottle for when you only feel like 1/2 a beer.
(There are top layers at the interface between the block and the model because the model is set to have top layers.)
[image]

santanig5, 2023-10-02:
Thanks Greg,
I did watch this YouTube (https://www.youtube.com/watch?v=0XNF3bGZw9E&t=456s)
[https://i.ytimg.com/vi/0XNF3bGZw9E/maxresdefault.jpg]<https://www.youtube.com/watch?v=0XNF3bGZw9E&t=456s>
How to 3D Print Nothing in Cura - Useful Slicer tip<https://www.youtube.com/watch?v=0XNF3bGZw9E&t=456s>
Ultimaker Cura is the most used 3D printing slicer, and has a set of tools that most people don't even know about. The Per-Model print settings enable you to...
www.youtube.com

However this gent used
Wall Thickness
Top/Bottom Thickness
Infill Density
which did not work for me.

How about a tool like S3D to show the model's cross section? I've always found this very valuable.

Kind Regards,
Gary

[image]<https://user-images.githubusercontent.com/64202104/272054009-a1ae4099-7e99-4ef5-94b3-3d6aacf1b2ed.png>

GregValiant, 2023-10-02:
The other settings should have worked.  "0" wall thickness is the same as a "0" wall count.  Same with top/bottom.
For technical reasons you would not be able to use "Infill Line Distance" set to some very large number and so you must use Infill Density set to "0".

santanig5, 2023-10-02:
Thanks again!