Ultimaker/Cura #16870: 5.4.0 "Slicing failed with unexpected error"

Cat-Lady, 2023-10-02:
### Cura Version

5.4.0

### Operating System

Debian Bookworm, AMD Van Gogh (Steam Deck)

### Printer

Ender 5 Pro

### Name abnormal settings

nope to all

### Describe model location

Model slices fine if rotated 90 degrees

### Describe your model

Model sliced fine before

### Add your .zip here ⬇️

[error.zip](https://github.com/Ultimaker/Cura/files/12787276/error.zip)

Cat-Lady, 2023-10-03:
The issue stops manifesting itself if "hole horizontal expansion" is set to "0". Also, the model slices very differently, if it is NOT set to zero, but model turned 90 degrees to allow slicing (part of model are just missing from the slice, like masts or some towers).

It is worth noting that this option should not affect masts and such non horizontally enclosed stuff.

GregValiant, 2023-10-03:
Thanks for the report.
Go to the Cura MarketPlace and download the "Mesh Tools" plugin.  It checks models for errors and that model has a lot of errors.
Cura will paint areas in polka-dots when errors in the model are encountered and a warning pops up as you see below.
[image]

Here is the report on that model from formware.co/OnLineSTLRepair
-> Reading file and indexing vertices
-> Analysed your file:
--> 297 Naked edges (?)
--> 10 Planar holes (?)
--> 10 Non-planar holes (?)
--> 5843 Non-manifold edges (?)
--> 4517 Inverted faces (?)
--> 12 Degenerate faces (?)
--> 5373 Duplicate faces (?)
--> 0 Disjoint shells (?)

Even for a boat that big that's a lot of errors.
Once the model is repaired you should get this...
[image]

I'm going to remove the bug label and go ahead and close this as it is a model problem.  I hope you understand.
You can also use MS 3D Builder to repair the model and there are other free on-line sites.

closed 2023-10-03 by GregValiant