Ultimaker/Cura #16872: Slicing fails when model is scaled to 150%

jake28mc, 2023-10-03:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Creality Ender-3 / Ender-3 v2

### Name abnormal settings

I've scaled the model to 150%. It works fine at other scales (140 and 160), but fails at 150%

### Describe model location

All default settings

### Describe your model

I used the plugin, my model is watertight. I've never printed my model before but others online have

### Add your .zip here ⬇️

[Basilisk.zip](https://github.com/Ultimaker/Cura/files/12788402/Basilisk.zip)

GregValiant, 2023-10-03:
Thanks for the report.
I'm starting to think that there is a conflict between Cura and some installations of Windows11.  There are a lot more reports regarding "won't slice" when the opsys is Win11.  I am on Win10Pro and I didn't have any trouble slicing the project file at 150%.

There are a couple of things you can try:

- Rotate the model in Cura
- Move the model away from the midpoint of the Cura build plate.
- Change the Initial Layer Height by a little bit.

Another possible workaround would be to turn off "Optimize wall printing order" and see if that makes a difference.

I'll mark this as a duplicate and the Cura Team will take a look.

jake28mc, 2023-10-04:
Strange issue - thankyou for your quick reply though!
Ended up slicing and printing at 149% lol - all three fixes you sent above also worked

Thankyou!

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0