Ultimaker/Cura #16876: Error en el corte

ftsskating, 2023-10-03:
### Cura Version

5.4.0

### Operating System

Windows11

### Printer

Ender 3 Custom

### Describe your model

El modelo es estanco. Lo he pasado por Meshmixer para asegurarme y esta en perfecto estado

### Add your .zip here ⬇️

[CE3E3V2_left side helmet DEADPOOL TEXTURED from Do3D (Cut 1).zip](https://github.com/Ultimaker/Cura/files/12794956/CE3E3V2_left.side.helmet.DEADPOOL.TEXTURED.from.Do3D.Cut.1.zip)

GregValiant, 2023-10-03:
Thanks for the report.
There is something about your "I3D-Tested PLA" setting profile that seems to be keeping this from slicing.  I switched to the "Standard" profile and selected "Keep Changes" and it sliced.

[image]

This seems to be a settings issue rather than a bug.  I'll leave the label on it for a bit and you can respond here.

ftsskating, 2023-10-04:
He cambiado al perfil estandar manteniendo los cambios, pero sige sin realizarme el corte. 
Tambien he probado a cargar el perfil estandar sin mantener los cambios del perfil personalizado, y tampoco me realiza el corte.

ftsskating, 2023-10-04:
A lo mejor es un problema con windows 11...

ftsskating, 2023-10-04:
Parece ser que con el perfil estandar que viene por defecto y recolocada la pieza, funciona correctamente. 
Algo del perfil personalizado esta creando conflicto. Revisaré a ver si doy con lo que es. 
Gracias!!!!

GregValiant, 2023-10-04:
Let's keep this open for now.  Hopefully you will come up with some indicator of where the conflict might be.  It may be something that the Cura team will want to address.

ftsskating, 2023-10-05:
Pues resulta que he podido realizar el corte pero solo funciona, si mantengo el perfil generico de Pla (Calidad estandar) y cambio las opciones a mano una por una a los parametros que necesito. 
Si trato de mantener los cambios, al cambiar del perfil personalizado (I3D-Tested PLA) al perfil generico de Pla(Calidad estandar) no funciona.

GregValiant, 2023-10-05:
That seems to indicate that there is indeed a setting conflict with the I3D profile, or it is somehow corrupt.
@ftsskating - When you created that custom profile - was it with 5.4.0 or with a previous version?

ftsskating, 2023-10-05:
Creo que lo cree en la version 5.3.0