Ultimaker/Cura #16877: Slicing Failed

DjamikTea, 2023-10-03:
### Cura Version

5.4.0

### Operating System

MacOS Sonoma (23A5337a)

### Printer

Anycubic i3 Mega

### Name abnormal settings

on standart settings, no sclicing 

### Describe model location

no

### Describe your model

no

### Add your .zip here ⬇️

[AI3M_amina_traf.3mf.zip](https://github.com/Ultimaker/Cura/files/12795058/AI3M_amina_traf.3mf.zip)

GregValiant, 2023-10-03:
Thanks for the report.
Your skirt settings are making the "footprint" of the model too big for the 210 x 210 build plate.  With three 0.4 wide lines all around, and an 8mm Skirt Distance all around the model takes up 221 x 221.

Make the following changes:

Skirt Line Count = 1 (that's all you need for a big model like this)
Skirt Distance = 2

This is a settings issue. so I'll go ahead and remove the bug label and close this.  I hope you understand.
[image]

closed 2023-10-03 by GregValiant

DjamikTea, 2023-10-14:
Oh, I completely forgot about the existence of a skirt, Thank you