Ultimaker/Cura #16879: Error en el corte

seis6seis, 2023-10-03:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Creality Ender-3

### Name abnormal settings

When I use an original Super Quality 0.12mm profile with 0.20 there is no error

### Describe model location

enters whole

### Describe your model

yes

### Add your .zip here ⬇️

[Link Proyect (File -> Save Proyect](https://we.tl/t-sjfCiYScSn?utm_campaign=TRN_TDL_05&utm_source=sendgrid&utm_medium=email&trk=TRN_TDL_05)

GregValiant, 2023-10-03:
Thanks for the report.
There is an often reported bug that can keep models from slicing.  It seems to be most prevalent with Win11.
Try Moving the models in Cura, Rotating the models, and changing the Initial Layer Height and the Layer Height.  Also, turning off "Optimize Wall Printing Order" can sometimes fool this bug.

I'll mark this as a duplicate and go ahead and close this as the Cura team has a lot of examples of this behavior.  I hope you understand.

This is with 5.4.0 on Win10 Pro.
[image]

closed 2023-10-03 by GregValiant

seis6seis, 2023-10-04:
Thank you very much, I have moved it on the Y axis a little more centered and it works. I don't know why it works with another height of capacity.

GregValiant, 2023-10-04:
I have no idea how the CuraEngine actually works but based on the fact that changing the initial X, Y, or Z often fools the bug; it's kind of like the "view" that the Engine has of the model makes a difference.