Ultimaker/Cura #16881: Mjolnir Mk6 Helmet single print.

ProximalAlpha, 2023-10-04:
### Cura Version

5.4.0

### Operating System

Win11

### Printer

Anycubic Chiron 0.6mm

### Name abnormal settings

Using Trees support instead of standard pillars

### Describe model location

Sliced when I used 0.6mm layers with a 0.4mm nozzle

### Describe your model

yes, 2 years ago with the same file just older Cura Version

### Add your .zip here ⬇️

[AC_1.17 triangles at 50percent.3mf.zip](https://github.com/Ultimaker/Cura/files/12798904/AC_1.17.triangles.at.50percent.3mf.zip)

This model was sized to be 182.+ mm between the ears at the neck.  It is tipped at approximately 68 degrees to fit in the build area.
The model is currently oriented at 45 degrees off-center and won't slice.  I got the file to slice at 0 degrees off-center with trees at 0.6mm layers with a 0.4mm nozzle.  [post processing won't be pretty... model thinks it is watertight, but reality says differently]

'why not just cut it into smaller pieces?'
"Because I know it will print in one solid piece on a Chiron, I've done it before. That's one of the only reasons I bought a Chiron."

Roasas, 2023-10-04:
the models too big for the plate my g

ProximalAlpha, 2023-10-05:
> the models too big for the plate my g

I don't know why it sliced this then. These photos are from a 0.4 nozzle at 0.6 layers (I forgot to change nozzle size in printer profile) this was taken at approx 7:10pm central (-5gmt)

[image]
[image]

This is the same model with 1.5mil+ tris.

closed 2024-04-03 by HellAholic