Ultimaker/Cura #16886: Thread from a bottle cap slicing incorrectly

Fabio3rs, 2023-10-04:
### Cura Version

5.4.0

### Operating System

Linux Mint 21.2

### Printer

Ender 3 Pro

### Reproduction steps

1. Add the .stl file in the Cura
2. Click on slice, tested in a lot of presets

### Actual results

In the preview tab, where is bottle cap thread are appearing full layers filling the cap area

### Expected results

Printing only where are the threads material

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]
[image]

The upload for the stl zipped is not working... the stl file https://drive.google.com/file/d/1J47c-mpsyx8yBApgjD0glu7QNsZTBdfz/view?usp=sharing

Asterchades, 2023-10-05:
Straight up I can tell you that this is a faulty model. Even a brief manual inspection shows superfluous geometry within the threads themselves and the upper part of the base is not connected to the walls, which tells me that this has been made of multiple pieces that have simply been put together - none of them having subsequently been connected to each other properly. It's actually broken to the point that 3D Builder takes over a minute to repair it on my R9 3900X and the end result looks like the faulty slice pictured.

A closer inspection shows this appears to be a model made of three parts - a hollow cylinder for the walls, a solid cylinder for the base, and the threads (which include their own additional wall). While the threads to appear to be attached to the walls at the top, being that the rest of their mesh is rotated slightly it means that there are dozens, if not hundreds of non-manifold segments created with walls continually crossing back and forth over each other.

Further to this there appears to be some kind of overly detailed mesh-like structure creating the rounded crest on the threads. I'm unsure if this is causing any problems directly, but the large amount of tiny (superfluous) detail is likely making the other issues even worse.

This model will require extensive repair at the source level in order to be usable. I would be extremely surprised if _any_ slicer can resolve it in its current state, and I don't feel like it would be reasonable to expect them to given that modelling tools with dedicated repair options can't make sense of it.

GregValiant, 2023-10-05:
Congratulations @Fabio3rs You have taken over first place in the UM Github "Who has the worst model" contest.
This is from formware.co/OnLineSTLRepair:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
**--> 730147 Non-manifold edges (?)
--> 419751 Inverted faces (?)**
--> 0 Degenerate faces (?)
-**-> 907854 Duplicate faces (?)**
--> 0 Disjoint shells (?)

There are 976396 triangles in the model and 2,057,752 errors.  The model volume is about 3300mm³.  That comes out to about 625 errors/mm³.  Frankly, I am in awe of the models "Error Density" which is a new unit of measurement I just made up.

I agree with @Asterchades that this model (as it is) is not repairable.

I'll go ahead and remove the bug label and close this.  There is too much wrong with the model to investigate further.  If the new model has issues this can be revisited or you can open a new report.  I hope you understand.