Ultimaker/Cura #16888: Switched from Draft mode to fine had a setting not saved in draft when I switched

petersg45, 2023-10-04:
### Cura Version

5.3.1

### Operating System

macos M1 Pro Ventura 13.4.1

### Printer

Anycubic

### Name abnormal settings

When I set scale to 20% error happens

### Describe your model

Genie Bottle from TinkerCad I created

### Add your .zip here ⬇️

[AV_JpGeniev4_FailedToSlice.3mf.zip](https://github.com/Ultimaker/Cura/files/12809174/AV_JpGeniev4_FailedToSlice.3mf.zip)
Happened after I scaled to 20% was fine at 25 and 30%

GregValiant, 2023-10-05:
Thanks for the report.
The title of the report indicates that the problem is a not being able to save settings, but the body of the report seems to indicate that the problem is with scaling the model.  Are there two issues?

petersg45, 2023-10-06:
I should have changed the title.  Seems to be the scaling that was the issue.

GregValiant, 2023-10-06:
Got it.
This is the standard 5.4.0 slicing bug.  I rotated the model 90° CCW and it sliced at 20%.  Other workarounds that can fool this bug are to move the model in Cura or change the initial layer height by a little bit.

I'll mark this as a duplicate and go ahead and close it as there are a lot of examples of this behavior.  Hopefully it will be fixed in 5.5.

The stock profiles are not "customizable".  You would need to create a custom profile based on a particular stock profile and then you can change whatever you want.

closed 2023-10-06 by GregValiant