Ultimaker/Cura #16889: unable to slice

Roasas, 2023-10-04:
### Cura Version

5.3.1

### Operating System

Mac

### Printer

Kobra Neo

### Describe your model

its a part of a mask, the other parts worked

### Add your .zip here ⬇️

[bottom half (4).stl.zip](https://github.com/Ultimaker/Cura/files/12810162/bottom.half.4.stl.zip)

Roasas, 2023-10-04:
[image]

GregValiant, 2023-10-05:
Thanks for the report and the file.
I used 5.4.0 as I don't have 5.3.1 loaded anymore.  It sliced fine, but the "Not Watertight" warning came up.  It would have been more helpful to have a project file.  I had to use my own setting profile and my printer (Ender 3 Pro) so I may be missing something here.

Here is the report on the model from [formware.co/OnlineStlRepair](https://formware.co/OnlineStlRepair):
-> Analysed your file:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
**--> 649 Non-manifold edges (?)
--> 868 Inverted faces (?)
--> 66 Degenerate faces (?)
--> 44 Duplicate faces (?)**
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 1347ms ------
-> Vertex count changed from 2720 to 3194 (+474)
-> Triangle count changed from 6756 to 7382 (+626)

Cura's MeshTools plugin (available in the MarketPlace) has some ability to repair models.  This one seems a bit extensive for it.  There are also on-line repair sites.  I like the formware site because of that report.

So what you may have here is a model problem rather than a bug.  There is a bug in 5.4.0 that can keep models from slicing and it was also present in 5.3.1.  You can try rotating the model in Cura, moving the model in Cura (although this one is pretty big), and changing the Initial Layer Height by a little bit.

I'm going to remove the bug label from this but I'll leave it open so you can respond.  The model has enough errors that there just isn't any way to tell if a bug is involved or if it is the model.  I hope you understand.

[image]

Roasas, 2023-10-05:
will this affect how the mack goes together?

GregValiant, 2023-10-05:
The physical size of the model doesn't change.  Some extremely small edges might change as the repair will close the open seams.  It's very doubtful that any changes made by a repair utility would be enough to affect the function or fit.  I've never seen that happen.

Roasas, 2023-10-07:
[image]
it still wont slice

GregValiant, 2023-10-07:
And I still don't have a project file to investigate.
I can't reproduce the problem with my setup because it slices fine for me.  Without a project file I can't do any more.

Roasas, 2023-10-08:
https://drive.google.com/file/d/1cn3MtCpLl07fNHUpygAGO6V1a0-ccYGx/view?usp=sharing
heres a drive link

Roasas, 2023-10-10:
nvm dont need it anymore

closed 2023-10-10 by Roasas