Ultimaker/Cura #16895: Changes in Cura 5.4.0 disabled multiselect files in Dialog File Selector

bugfinder70, 2023-10-05:
### Cura Version

5.4.0

### Operating System

Windows 10 22H2

### Printer

Creality Ender 3 S1 Pro

### Reproduction steps

Open valid files *.stl *.3mf: Want to select 10 files.
But in 5.4.0 I can only select 1 file.
I remember: in the past there is the possibility to select many files in the open dialog.

### Actual results

Only 1 file is selectable in the open files dialog box

### Expected results

Many files in the open diallog box selectable

### Add your .zip and screenshots here ⬇️

nothing

LisaLisa480, 2023-10-05:
I am on Mac, Ventura 13.4 (22F66), Cura Version 4.5.0 and same for me. Dragging and dropping multiple files works, just not through the open files dialog box.

GregValiant, 2023-10-05:
Thanks.
This is a duplicate of #15879 .  I'll mark it and go ahead and close this.  It should be fixed in the next release.  I hope you understand.

closed 2023-10-05 by GregValiant