Ultimaker/Cura #16900: Create a setting "Max Flow"

FabioM3, 2023-10-06:
### Is your feature request related to a problem?

To achieve maximum performance, a setting so that the Cura does not exceed a certain "Maximum Flow" will make prints faster, and we don't need to reduce the speed.

Because in the end it's the nozzle's maximum Flow that matters when it comes to print time.

### Describe the solution you'd like

An checkbox "Use Max Flow"
An textbox "Max Flow"

The printer will not go above the max flow defined, showing it in the Speed and Flow views too.

### Describe alternatives you've considered

An alternative is that Cura calculates differently when Variable Layer Heigth are used. Or have an checkbox to calculate differently.

### Affected users and/or printers

All.

5axes, 2023-11-08:
This is normaly the goal of the Parameter "Adaptative Layers Maximum Variation"  to limit the height of the layer.   Then the  maximum Flow is just the  Maximum Layers Height X Line Width X Print Speed

onolox, 2023-11-08:
You can only adjust the printing for thicker lines. Thinner lines have less flow and can slow down the print.
By using this setting, I can adjust for thinner layers and cura will prevent the printer from exceeding the maximum flow.

FabioM3, 2023-11-08:
You can only adjust the printing for thicker lines. Thinner lines have less flow and can slow down the print.
By using this setting, I can adjust for thinner layers and cura will prevent the printer from exceeding the maximum flow.
I borrowed this idea from Prusa.