Ultimaker/Cura #16901: Failed slicing of lightsaber screw

Pissman2020, 2023-10-06:
### Cura Version

5.4.0

### Operating System

win 10

### Printer

pissman blue

### Name abnormal settings

turning off tree supports relieves the error. Normal supports works.

### Describe model location

center of build plate, default orientation

### Describe your model

First time trying to slice, mesh fix didn't work

### Add your .zip here ⬇️

[Qui-Gon Jinn Lightsaber-Covertec_Screw-silver.zip](https://github.com/Ultimaker/Cura/files/12832928/Qui-Gon.Jinn.Lightsaber-Covertec_Screw-silver.zip)

Pissman2020, 2023-10-06:
This is bottom of the barrel importance because I found a way to get it to slice, but no idea why it wouldn't slice with those settings or would effect models that need those settings on.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic