Ultimaker/Cura #16902: Slicer error at simple things that work before

EmEiBee, 2023-10-06:
### Cura Version

5.4.0

### Operating System

Windowss 10 pro

### Printer

Modificated GeeeTech I3proC

### Name abnormal settings

no

### Describe model location

no

### Describe your model

Simple Tools for lubricate linears

### Add your .zip here ⬇️

[GI_M4NutCap_1.zip](https://github.com/Ultimaker/Cura/files/12833461/GI_M4NutCap_1.zip)

GregValiant, 2023-10-06:
Thanks for the report.  This is a strange one.
I tried all my tricks and I can't get it to slice if your Geeetech is the active printer.
I can't get it to "Not Slice" if my Ender is the active printer.

The Cura team will take a look.

EmEiBee, 2023-10-06:
jepp, that's crazy. If I only use one part (just right now testing it) it will slice without problems. But if it more then one part, it will not slice anymore.
But I'm completely sure that it once worked. I bet my head on it...

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic