Ultimaker/Cura #16905: I can't slice any models i try to slice

Oneill67, 2023-10-07:
### Cura Version

5.4.0

### Operating System

windows 10

### Printer

Ender3 V2

### Name abnormal settings

No change since the last time

### Describe model location

I have the issue with all models i tried to slice

### Describe your model

Yes, It's a models i've downloaded from Thingiverse

### Add your .zip here ⬇️

[CE3E3V2_Stand_v2.zip](https://github.com/Ultimaker/Cura/files/12837437/CE3E3V2_Stand_v2.zip)

GregValiant, 2023-10-07:
Thanks for the report.
I didn't have any trouble slicing the project.
Are you sure you provided the project file with the settings you want to use and the model in the orientation you want to print it in?
It will fail with your current setup.
[image]

Oneill67, 2023-10-07:
I've found the solution, it's the Antivirus software Avast. I stoped the protection and I can slice the project.
To resolve the problem, add an exception Files/Directory in to Avast on the "C:\Program Files\UltiMaker Cura 5.4.0\*".
[image]

GregValiant, 2023-10-09:
It appears that it wasn't a Cura problem.  Can we go ahead and close this?

Oneill67, 2023-10-10:
Yes you can close the issue

Oneill67, 2023-10-10:
Yes, it can be closed

closed 2023-10-10 by Oneill67