Ultimaker/Cura #16908: Slicing failed with thin 3D image

txyankee2, 2023-10-07:
### Cura Version

Ultimaker Cura 5.4.0 - see attachment

### Operating System

Windows 11, Processor	Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz, 2001 Mhz, 4 Core(s), 8 Logical Processor(s)

### Printer

Mingda Magician X2

### Name abnormal settings

See attached project file

### Describe model location

See attached screenshot

### Describe your model

Converted a 2D image to a 3D image using imagetostl.com

### Add your .zip here ⬇️

[image]
[image]
[Zipped-projectandSTLfile.zip](https://github.com/Ultimaker/Cura/files/12838465/Zipped-projectandSTLfile.zip)

GregValiant, 2023-10-07:
Thanks for the report.
The model is not watertight so it won't slice correctly.  Here is a report on the model.
**--> 1546 Naked edges (?)
--> 2 Planar holes (?)**
--> 0 Non-planar holes (?)
--> 0 Non-manifold edges (?)
--> 0 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 107993ms ------
-> Vertex count changed from 102977 to 107035 **(+4058)**
-> Triangle count changed from 204412 to 214066 **(+9654)**

Did you post a bug report on the "imagetostl.com" site?  That litho seems to be very poor.

This is the repaired model sliced in 5.4.0.

txyankee2, 2023-10-10:
Hello Greg,

Did you mean to send me the repaired file? I did not have an attachment on the email.  I don't see an attachment on the case.

Please send the repaired file.

Thanks for your help.

Gene Burnell

txyankee2, 2023-10-10:
Hello Greg,

Sorry.  I think I see the attachment on the case.  It is named zipped-protectandSTLfile.

Thanks,
Gene Burnell

txyankee2, 2023-10-10:
Hello Greg,

No, that attachment I found it the one that I posted to case.

Please send me the repaired file.

Thanks,
Gene Burnell

GregValiant, 2023-10-10:
I'm sorry, but I never do peoples homework for them.

txyankee2, 2023-10-10:
Greg,

If that’s the case, the details of your analysis mean absolutely nothing to me. It’s too technical for a novice 3D hobbyist.

How do I fix it so it will slice?

Gene Burnell

GregValiant, 2023-10-10:
I'm a retired fireman.  All I know about this stuff I learned from researching and finding out how to do things since I bought a printer because I would be home a lot during Covid.
I won't be there to hold your hand for the next one so you need to learn how to do these things yourself.  Google is your friend.  Do what I did and search for "STL repair".  One of the things you will come across is MS 3D Builder.  There are also free on-line sites like the one I used to generate that report.  All you need to notice about that report is that "0" errors is a good thing, and "Lots of errors" is a bad thing.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic