Ultimaker/Cura #16909: Slicing failed with new model

citationfour, 2023-10-07:
### Cura Version

5.3.1

### Operating System

Win11

### Printer

Custom

### Describe your model

Not sliced before.

### Add your .zip here ⬇️

[CFFFP_base.zip](https://github.com/Ultimaker/Cura/files/12838833/CFFFP_base.zip)

GregValiant, 2023-10-07:
Thanks for the report.
The model is 76 x 76.  The printer is 80 x 80.  That leaves a 2mm border all around the model for your skirt to fit in.  Your skirt distance is 3.0 so slicing fails as none of the skirt can fit on the bed.
Turn off the bed adhesion or set the skirt distance to 1.0.

I'll go ahead and remove the bug label and close this as it is a settings issue.  I hope you understand. 

Skirt distance at 1.0.
[image]

closed 2023-10-07 by GregValiant

citationfour, 2023-10-12:
cheers. didn't realise that