Ultimaker/Cura #16910: First Layer Settings not applied on print by part

Ingegneus, 2023-10-07:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Anycubic Mega X Knutwurst Marlin release

### Reproduction steps

1. insert multiple objects to print
2. set print sequence to one at a time
3. set for example first layer bed temperature to 40°C and normal bed temperature to 30°C  
4. print

### Actual results

for the first print object the bed heats up to 40°C but for the subsequent it doesn't

### Expected results

that every first  layer of every print objects gets treated the same. this is very important for bed adhesion

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-10-08:
Thanks for the report.
This is a duplicate of #15566 .  In that report I describe a workaround (at least for temperature).  See if it helps.
I'll mark this as a duplicate and I'll leave it open so you can respond.

GregValiant, 2023-10-08:
I just looked at another report regarding "One at a Time" and my gcode shows this:

G1 F2400 E3822.136
M107
G0 F3000 X97.246 Y168.625 Z25
G0 X95 Y28.078
**M140 S65
M105
M190 S65
M104 S210
M105
M109 S210**
;LAYER_COUNT:100
;LAYER:0

So with the setup I used there it appears that the temperatures are being reset as they should be.
We will need a project file to investigate further.

Ingegneus, 2023-10-09:
here is the zipped project file. let me know if you need anything else
[ornament.zip](https://github.com/Ultimaker/Cura/files/12847312/ornament.zip)

GregValiant, 2023-10-09:
Thanks.
I can't find the problem but it is somehow related to your printer definition.
If I change to my Ender 3 Pro there is no problem.
If I change to an AnyCubic i3 there is no problem.
The difference seems to be when an Anycubic printer "inherits" the "anycubic_i3_mega_s.def.json" definition and your printer does, the ones that work do not inherit that "mega_s" file.
As I was trying to figure this out:  when I would switch between AnyCubic printers they would come up with the wrong build surface size in Cura.

I'm going to leave the bug label on this for now and maybe someone else here will take a look.  It sure seems to be a "definition" problem.  The definition files come from members of the community or from the printer manufacturer so in that case it would not be a UltiMaker Cura bug.

This is from the gcode generated when your Mega X is the active printer.  The only thing that gets reset for Layer:0 is the fan.
;TIME_ELAPSED:813.962343
G1 F1800 E390.22758
M107
G0 F7200 X93.739 Y160.355 Z7.6
G0 X136.502 Y116.501
;LAYER_COUNT:15
;LAYER:0

All I did here was switch printers to the i3 Mega.  The temperature settings showed up.
;TIME_ELAPSED:1362.556933
G1 F2400 E806.47766
M107
G0 F3000 X1.274 Y95.124 Z7.6
G0 X91.502 Y71.501
M140 S30
M105
M190 S30
M104 S200
M105
M109 S200
;LAYER_COUNT:15
;LAYER:0

Ingegneus, 2023-11-02:
I updated to 5.5.0 and it seems to be working now. I would leave it open for now in case that something pops up. Or would you like me to close this?

GregValiant, 2023-11-02:
I'll go ahead and close this then.  The duplicate15566 is still open.

closed 2023-11-02 by GregValiant