Ultimaker/Cura #16917: [5.4.0] Not using shortest route between areas to bridge

techflick, 2023-10-09:
### Cura Version

5.4.0

### Operating System

Windows 10, 22H2, 19045.3448

### Printer

Anycubic i3 Mega and Chiron

### Reproduction steps

I wanted to force Cura to use bridging over narrow gaps on my model (by bumping up Bridge Skin Support Threshold), so bending of the bridged area would be more reliable.

### Actual results

I then discovered that Curas bridging is not (in all areas) following the shortest path between the areas to bridge /across the area to bridge, which leads Cura to use irrationally long bridges and in 90° to the expected direction over the intended bridging area.
[image]
[image]
[image]

### Expected results

1. I expected Short bridges, only across the narrow gap or at least over the whole area, in the same direction as the expected short bridges.
[image]
2. After some consideration I concluded it must be because the length between the infill and the outer wall (a) is considered shorter than the path directly across the area to bridge (d) / it is ignored that to bridge the whole path directly across the area (b) would still be much shorter and more reliable than across the whole width (c) of the part. Depending on how the algorithm is working or is intended to work.

### Add your .zip and screenshots here ⬇️

[irrational Bridging .zip](https://github.com/Ultimaker/Cura/files/12844072/irrational.Bridging.zip)

tripletto, 2023-10-11:
Jupp, had this as well. Cura 5.4 really loves to bridge 100 mm in length when it had the option to bridge only 20 mm instead, which would be the rational choice.

techflick, 2023-10-12:
While I was compiling the simple model for reproducing of the bug I actually discovered as a workaround that it is working as intended when there is a bit more meat on the narrow side. But some times it is not possible to alter the model in this way and so the question stays: In the case of the narrow area, why are the short sides (brown) more valid then the long side (blue) to bridge between, if one of this possibility does not bridge the shortest path over the gap?
[image]
[More Meat.zip](https://github.com/Ultimaker/Cura/files/12878123/More.Meat.zip)

techflick, 2024-05-10:
I had a situation with the same problematic case and it seems 7.5.1 is working as expected for now.

gsec0, 2024-10-27:
In case anyone else is googling this and looking for a solution, as of Cura 5.8.0 there is an option called "Bridge sparse infill max density" 
[image]

I started decreasing this value until I got the desired result, the threshold for my print was 9%, this will depend on many factors.
[image]

Basically as I understand it, If the walls below the first bridge layer are very thin (for example you have a slope, see pic) 
[image]
Then the bridging algorithm will not want to bridge there as it thinks there isnt enough supporting material (only 1 or 2 layers). So decreasing this threshold allows the algorithm to use those thinner walls below the first bridge layer as support material.

I hope that explains it