Ultimaker/Cura #16926: Error slicing model in Cura 5.4.0

Juanfminarro, 2023-10-10:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Ultimaker S5

### Describe your model

No. No.

### Add your .zip here ⬇️

[UMS5_INFERIOR_DERECHA_MODIF_FINAL_9_PLUS.zip](https://github.com/Ultimaker/Cura/files/12856198/UMS5_INFERIOR_DERECHA_MODIF_FINAL_9_PLUS.zip)

GregValiant, 2023-10-10:
Thanks for the report.
I couldn't get it to slice without fooling with it.  I finally got it to slice by setting the Initial Layer Height to 0.19 and the Layer Height to 0.21.  I'll mark this as a duplicate report and the Cura team will take a look.

Models that have spires (knife edge features or features that come to a point) often will not slice correctly.  As the width of a feature approaches zero, the "Line Width" will not longer fit (even for just a dot of plastic).  Almost all of the high points of this model will not print.  It will however end up as a LOT of islands that will require a LOT of retractions.
[image]

You can see that some of the spires aren't going to do well.  I think supports "Touching Build Plate" would be fine for this.
[image]

Juanfminarro, 2023-10-10:
Good afternoon.

Strange error, because it only appears if I scale the piece. The segmentation is normal if done at 100%, but not at 466% which is what I need to print.

And on the other hand, I have tested it with version 4.12.1 and it segments it without any problem with any type of % scale.

This makes me think it may be a defect in Cura version 5.4.0

Greetings.

[cid:e26b5236-51cc-4e99-8a66-5657b481d78f]

[image]<https://urldefense.com/v3/__https://user-images.githubusercontent.com/64202104/273881775-564bba52-ccf8-4bd4-b23f-a817598cdcb1.png__;!!D9dNQwwGXtA!TmQoVkpI7B1TkXEcIPn3_z2EVY3eeLZQplR_WtRk0ea8aM1jl6Mp9iFIxzTGGZBuTIogzX-nU0eoLHLTFy9Y$>

[image]<https://urldefense.com/v3/__https://user-images.githubusercontent.com/64202104/273891157-6892b1ce-266f-43fe-ad23-d481309eded7.png__;!!D9dNQwwGXtA!TmQoVkpI7B1TkXEcIPn3_z2EVY3eeLZQplR_WtRk0ea8aM1jl6Mp9iFIxzTGGZBuTIogzX-nU0eoLO5ouMnv$>

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0