Ultimaker/Cura #16932: UltiMaker Cura 5.5.0-beta.1 Don't Open in Linux

erickwill, 2023-10-10:
### Cura Version

5.5.0-beta.1

### Operating System

Linux - Arch Linux Kernel 6.5.6-arch2-1

### Printer

Tronxy X5SA

### Reproduction steps

It does not open the app when the AppImage file is executed. In the terminal, when executed through there I got this message: 
`[erick@mobile-hp Apps]$ ./UltiMaker-Cura-5.5.0-beta.1-linux-X64.AppImage 
PyInstaller/loader/pyimod02_importers.py:352: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
libEGL fatal: DRI driver not from this Mesa build ('23.0.4-0ubuntu1~22.04.1' vs '23.2.1-arch1.2')
`

### Actual results

After executing the appimage, failing, got this error:
`PyInstaller/loader/pyimod02_importers.py:352: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
libEGL fatal: DRI driver not from this Mesa build ('23.0.4-0ubuntu1~22.04.1' vs '23.2.1-arch1.2')
`

### Expected results

To execute the appimage and open it successfully.

### Add your .zip and screenshots here ⬇️

The file I used is the last in git: UltiMaker-Cura-5.5.0-beta.1-linux-X64.AppImage

jellespijker, 2023-10-10:
do you have libfuse2 installed on your system? That is needed to run AppImages in general

erickwill, 2023-10-11:
The last stable version works without any additional library. Does it's necessary from now on?

erickwill, 2023-10-11:
@jellespijker I didn't find it in the Arch Linux repository. Can you confirm if it'll be necessary from now on and a place to download and install it? Thanks

erickwill, 2023-10-11:
@jellespijker 

As you can see I'm able to use the last stable version without any other lib. 

Though, I have a few "issues" under the hood: 
`[erick@mobile-hp Apps]$ ./UltiMaker-Cura-5.4.0-linux-modern.AppImage 
PyInstaller/loader/pyimod02_importers.py:352: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///tmp/.mount_UltiMaCrdbL4/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltiMaCrdbL4/share/cura/resources/qml/Actions.qml"
file:///home/erick/.local/share/cura/5.4/plugins/PowerCost/PowerCost/PowerCostWidget.qml:13:2: QML PowerCostWidget: Binding loop detected for property "height"
file:///home/erick/.local/share/cura/5.4/plugins/PowerCost/PowerCost/PowerCostWidget.qml:13:2: QML PowerCostWidget: Binding loop detected for property "height"
file:///home/erick/.local/share/cura/5.4/plugins/PowerCost/PowerCost/PowerCostWidget.qml:12:2: QML PowerCostWidget: Binding loop detected for property "width"
file:///tmp/.mount_UltiMaCrdbL4/share/cura/resources/qml/PrintSetupSelector/Recommended/RecommendedPrintSetup.qml:96:17: QML QQuickItem: Binding loop detected for property "height"
file:///tmp/.mount_UltiMaCrdbL4/share/cura/resources/qml/PrintSetupSelector/Recommended/RecommendedPrintSetup.qml:96:17: QML QQuickItem: Binding loop detected for property "height"
`
[image]

jellespijker, 2023-10-11:
@erickwill can you try to run this version to see if that solves your issue: https://github.com/Ultimaker/Cura/suites/17114831102/artifacts/977675757 

This version is based on 5.5.0 but it will have a version 5.6.0-alpha designation. So it is best if you clean up the local cura/5.6 folder afterwards: `rm -rf ~/.local/share/cura/5.6` should do the trick.

Gert-Boers, 2023-10-13:
I've downloaded the 5.5.0 beta today and get the same error message on Arcolinux:
libEGL fatal: DRI driver not from this Mesa build ('23.0.4-0ubuntu1~22.04.1' vs '23.2.1-arch1.2')

So, I can confirm 5.5.0 beta does not run on Arch atm.
And, like erickwill, on my system 5.4.0 also runs without issues.

jellespijker, 2023-10-13:
@Gert-Boers please the the new AppImage which I shared one post up

Gert-Boers, 2023-10-13:
@jellespijker 
I tried downloading that before posting, but I got an 404 webpage. But now I can download (?).

This one (5.6.0) crashes on me with a Gui:
[image]

erickwill, 2023-10-13:
> @erickwill can you try to run this version to see if that solves your issue: [UltiMaker-Cura-5.6.0-alpha+cura_11146_d8ef4c-linux-X64-AppImage](https://github.com/Ultimaker/Cura/suites/17114831102/artifacts/977675757)
> 
> This version is based on 5.5.0 but it will have a version 5.6.0-alpha designation. So it is best if you clean up the local cura/5.6 folder afterwards: `rm -rf ~/.local/share/cura/5.6` should do the trick.

Works perfectly. Sorry for the delay.

erickwill, 2023-10-14:
> @erickwill can you try to run this version to see if that solves your issue: [...]

Just out of curiosity, I wonder if there is a nightly for the 5.6.0? If yes, is it open for public testers? Thanks for your hard work!

MariMakes, 2023-10-16:
We've added a ticket to the backlog with the intent to improve this, hopefully soon. 
For internal reference CURA-11132

Thanks for the report! 👍

didli, 2023-10-16:
For infos, this alpha version (UltiMaker-Cura-5.6.0-alpha+cura_11146_d8ef4c-linux-X64-AppImage) crashed for me with the exact same log previously given in my [report](https://github.com/Ultimaker/Cura/issues/16992) :
```log
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Papirus-Dark/16x16/actions/image-missing.svg: Unable to load image-loading module: libpixbufloader-svg.so: libpixbufloader-svg.so: cannot open shared object file: No such file or directory (gdk-pixbuf-error-quark, 5)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Papirus-Dark/16x16/actions/image-missing.svg: Unable to load image-loading module: libpixbufloader-svg.so: libpixbufloader-svg.so: cannot open shared object file: No such file or directory (gdk-pixbuf-error-quark, 5)
```

jellespijker, 2023-10-24:
Should be fixed in the stable release with the merge of #17080

closed 2023-10-27 by saumyaj3

chaserhkj, 2023-11-15:
Unfortunately, this issue still exists after #17080. I have tried both 5.5.0 release and the nighty build from github actions. They all behave in the same way of crashing once Cura tries to show a file open dialog.

I have xdg-desktop-portal properly installed and configured. Can confirm they work by testing in flatpak apps.

The only workaround that I could get Cura to work is still by adding `-platformtheme gtk3`  in command line options as suggested by some people in #12266

Jorengarenar, 2023-12-02:
@saumyaj3 @MariMakes 

Well, I've just installed the release 5.6.0 and the issue remains the same,
with the same workaround of launching with `-platformtheme gtk3`

Gert-Boers, 2023-12-02:
What is a bit strange: I installed cura-5-bin from the AUR and it downloads exactly the same AppImage.
But when I run that, it's working perfectly...

dcervenkov, 2023-12-30:
@Gert-Boers and others who might stumble on this; starting the AppImage for Cura 5.5 and 5.6 from the command line with `QT_STYLE_OVERRIDE=""` causes it to work for me. This is not needed for Cura 5.4. This is one of the things that the cura-5-bin package from the AUR does. 

Funnily, the AUR package has a typo `QT_QPA_PLATAFORM` instead of `QT_QPA_PLATFORM`, but since it is not needed, it's not a big deal.

Gert-Boers, 2023-12-30:
@dcervenkov Thanks!
That indeed does work. But as the AUR also installed 5.6.0, I'll most likely use that. 
But it's good to know, for when future updates give the same problem.

acornejo, 2024-03-28:
Not sure why this is marked as fixed, I run into this issue on Linux when running the latest available version on the website. the workarounds suggested here do work though.

Estebiu, 2024-04-01:
Currently having the same problem on arch linux.

Running cura with '-platformtheme gtk3' as mentioned above resolves the crashing issue, but the files still aren't imported into cura. Nothing happens. 

Is there a workaround?