Ultimaker/Cura #16935: Manjaro Linux: App doesn't load

joe8mofo, 2023-10-11:
### Cura Version

5.5 Beta

### Operating System

23.1.9-manjaro

### Printer

Ender 3 Pro

### Reproduction steps

I attempted to run the app. It didn't load. So I attempted to load it from terminal.

### Actual results

The following error is seen:
    ~/Desktop  ./UltiMaker-Cura-5.5.0-beta.1-linux-X64.appimage                                       ✔ 
PyInstaller/loader/pyimod02_importers.py:352: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
libEGL fatal: DRI driver not from this Mesa build ('23.0.4-0ubuntu1~22.04.1' vs '23.1.9-manjaro1.1')

    ~/Desktop     

### Expected results

I expect the application to load.

### Add your .zip and screenshots here ⬇️

None

echow2001, 2023-10-11:
same issue 

`➜  Applications ./UltiMaker-Cura-5.5.0-beta.1-linux-X64.AppImage
/usr/bin/AppImageLauncher: /lib64/libcurl.so.4: no version information available (required by /usr/bin/../lib/x86_64-linux-gnu/appimagelauncher/libappimageupdate.so)
QSocketNotifier: Can only be used with threads started with QThread
PyInstaller/loader/pyimod02_importers.py:352: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
libEGL fatal: DRI driver not from this Mesa build ('23.0.4-0ubuntu1~22.04.1' vs '23.2.1')

➜  Applications 
`

jellespijker, 2023-10-11:
@echow2001 and @joe8mofo can you try to run this version to see if that solves your issue: [UltiMaker-Cura-5.6.0-alpha+cura_11146_d8ef4c-linux-X64-AppImage](https://github.com/Ultimaker/Cura/suites/17114831102/artifacts/977675757)

This version is based on 5.5.0 but it will have a version 5.6.0-alpha designation. So it is best if you clean up the local cura/5.6 folder afterwards: `rm -rf ~/.local/share/cura/5.6` should do the trick.

joe8mofo, 2023-10-11:
5.6.0-alpha loads correctly.