Ultimaker/Cura #16936: no longer slicing anything

j-snijder, 2023-10-11:
### Cura Version

5.5.0 beta 1

### Operating System

windows 11

### Printer

ender 3 pro

### Name abnormal settings

nothing is sliced anymore, NOTHING

### Describe model location

standard

### Describe your model

benchy

### Add your .zip here ⬇️

[CE3PRO_jungle.zip](https://github.com/Ultimaker/Cura/files/12866458/CE3PRO_jungle.zip)

j-snijder, 2023-10-11:
Want to add that all models I tried slide perfectly in Cura 5.3.1 and all printer, material and other settings are all the same, nothing changed or different in whatever settings are there.

The problem started with versions after v 5.3.1 and persist

j-snijder, 2023-10-11:
As soon as I am free tonight I'll try and send you all logs from v4.5.x and the project and logs from v5.3.1 to compare.
Thus bug must be found, its getting worse and I am not the only one that has this problem 🥴

j-snijder, 2023-10-12:
[3 times cura to compare.zip](https://github.com/Ultimaker/Cura/files/12879222/3.times.cura.to.compare.zip)
Sorry for the delay, as promised I send you a few projects and logs to compare for why slicing is not working.
Strange enough v 5.4.0 is slicing half of the time and now v 5.5.0 does not slice any of my prints, really not even the simplest ones.
I use the same config on all versions ( kept 2 versions because I already had problems with 5.4.0) and now I wanted to see if in 5.5.0 the problem was adressed its getting worse.

MariMakes, 2023-10-12:
Hey @j-snijder,

I'm seeing this error:  Backend exited abnormally with return code 2!
But also Backend exited abnormally with return code 1!
That's usually something that the team can help troubleshoot. I'll bring it up. 
Fingers crossed 🤞 

Are you very attached to your printers, profiles and materials? 
I think the proposal might be to do a clean install, and without a proper back-up you might lose those.

j-snijder, 2023-10-14:
I have backups through cura, but when I set them back into a clean install the problem is back again. Done that already.

MariMakes, 2023-10-16:
Hi @j-snijder,

I had one of the developers take a look. They mention that return 2 seems to indicate that the installation isn’t clean, and return 1 can signal FPE ('Floating Point Error'), which are divisions by zero and the like.

Could you try removing some old installs and installing a fresh MSI? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1

Let's hope that resolves your issue 🤞

j-snijder, 2023-10-17:
Hi,As I am on a short vacation at the moment, and abroad, I can not right now.As soon as I get back I'll try to cleanup and reinstall.Greets, Jan

Jan-Snijder, 2023-12-24:
SOLVED ! damaged ? file in configuration files.