Ultimaker/Cura #16938: Target machine name in line 2 of g-code prevents Ender-3 reading time/filament use/layer height

Slashee-the-Cow, 2023-10-11:
### Cura Version

5.5 Beta 1

### Operating System

Windows 10 x64 22H2

### Printer

Ender-3 / Ender-3 v2

### Reproduction steps

Note: I usually use a custom profile I made for my Ender-3 V3 SE, but for this test I switched to the stock Ender-3 / Ender-3 v2 profile to be easily reproduced by other people. I've confirmed that the problem appears in the files for both my E3V3SE profile and the stock E3/E3v2 profile.

Slice a model (really, anything will do) in both 5.4 and 5.5 beta.

I used the project file included in the zip file (the fact that it's tiny doesn't matter, I tried more complex models and the same thing happened). Set it to the stock E3/E3v2 profile in Cura and standard quality preset.
I sliced it in both 5.4 and 5.5 beta 1 and saved separate g-code files.

Also I copied the file sliced in 5.5 beta and removed the second line (the comment starting in `;TARGET_MACHINE.NAME:`) manually.

### Actual results

- On the printer's screen, before the print, it displays the estimated time, filament usage and layer height when choosing to print from the file sliced in 5.4.
- The same information appears when using the manually modified file from 5.5 beta 1.
- When selecting the file sliced in 5.5 beta 1, estimated time, filament usage and layer height all appear as 0.

### Expected results

Estimated time, filament usage and layer height should be read correctly by printer regardless of Cura version used to slice model

### Add your .zip and screenshots here ⬇️

[e3 no info little disc.zip](https://github.com/Ultimaker/Cura/files/12868367/e3.no.info.little.disc.zip)

MariMakes, 2023-10-12:
Hey @Slashee-the-Cow,

Welcome to the UltiMaker Cura Github 🚀 
I believe I recognize you from the UltiMaker Community Forum!

The Target Machine Name has been introduced here https://github.com/Ultimaker/CuraEngine/pull/1871
and we believed it was harmless since it was just a comment. 

I'll bring it up to the team to see what they can do to improve it.
Fingers crossed 🤞

Slashee-the-Cow, 2023-10-13:
World's most simple post processor as a workaround:
[RemoveTargetMachine.zip](https://github.com/Ultimaker/Cura/files/12888086/RemoveTargetMachine.zip)

MariMakes, 2023-10-16:
@smartin015 we've had this report with the Target Machine Name causing incorrect times/materials on the display. Do you need the comment to be this high? Could it be lower in the list of comments?

@GregValiant, you have an Ender-3, right? Is it something you are seeing as well?

GregValiant, 2023-10-16:
Yes I do and no I'm not.  I have the 8 bit board with the older 1.1.8 firmware.
My guess here is that the newer firmware is picking the second and third lines of data[0] rather than looking for ";TIME:" and ";Filament used:".  This is likely for compatibility with the "Creality Slicer".  This problem will also show up with the proposed "DisplayInfoOnLCD" post processor.  I'll have to change that to move a couple of insertions below the "filament used" line.

The machine name could be added to the end of the "generated with" line.  It would be out of the way there. 

;POSTPROCESED
;  [RemoveTargetMachineV2]
;Generated with Cura_SteamEngine 5.5.0-beta.1 || For Printer: Creality Ender 3 Pro

[RemoveTargetMachineV2.zip](https://github.com/Ultimaker/Cura/files/12920003/RemoveTargetMachineV2.zip)

smartin015, 2023-10-16:
And here I thought a comment-only change would be safe :-\ it's unfortunate the firmware doesn't actually parse the headers to understand the values that it's reading. IMO this should be fixed on the reader side so it's more resilient to such changes. 

It'd be fine to move it down a ways, but I'd like folks to all agree on the notion that named comments in the sliced file should be strictly ordered like this. It seems quite brittle to make that assumption.

wawanbreton, 2023-10-23:
Hi @Slashee-the-Cow, I made a quick fix to put the newly added comment at the end of the header. This will be added in 5.5 stable.

However, this is not a healthy behavior and I would like to make sure the Ender firmware can handle such changes in the future. Could you give me your exact firmware version ?

Slashee-the-Cow, 2023-10-26:
Sorry for the delayed response, got a lot of things on the boiler at the moment. Right now I'm running 1.0.4. 1.0.6 just came out and I'm debating installing it because 1.0.4 works :)

GregValiant, 2023-10-26:
The "machine_name" was moved to the end of data[0] in 5.5.0.  That would seem to indicate it is no longer an issue.
I'll remove the bug label as this was actually (poor) expected behavior.  Should I leave this open or can it be closed?

Slashee-the-Cow, 2023-10-26:
In the middle of a big print now but if you give me fifteen hours I can disable my post which removes that line and double check on my printer.

Slashee-the-Cow, 2023-10-26:
Behaviour is fixed on my printer.

closed 2023-10-26 by Slashee-the-Cow

wawanbreton, 2023-10-30:
From what I could see, the Marlin-based Creality firmwares should handle the header properly, whatever the order.

However, for this printer the firmware seems to be different, and closed-source. So the only thing we could do is kindly ask them to make this piece of code a bit stronger, but I don't know how responsive they are to such requests.

Slashee-the-Cow, 2023-10-31:
Can confirm that behaviour is unchanged in firmware version 1.0.6.