Ultimaker/Cura #16941: First run crash

MikeOxawopper, 2023-10-11:
### Cura Version

5.5

### Operating System

windows 10

### Printer

ender 3 v2

### Reproduction steps

1. install software
2. open software

### Actual results

CRASH
[image]

### Expected results

Be nice if it at least opened

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/Ultimaker/Cura/files/12869219/cura.log)

lonlonranchdressing, 2023-10-11:
Just chiming in, Same issue. Any workarounds?

MikeOxawopper, 2023-10-11:
I bodged my way around it.
i hit the regenerate and backup button at the bottom

then opened 5.4 and 5.5 folders side by side by using the show folder option within 5.4 help menu.

i then opened each folder in turn in 5.4 copied the data within the folders to the matching folder in 5.5

you have to create a new printer but once in i just deleted that

MariMakes, 2023-10-12:
@5axes, I believe this is the issue [you've reported in the UltiMaker Community](https://community.ultimaker.com/topic/44294-ultimaker-cura-55-beta-released/?do=findComment&comment=326232).

Do you agree?

5axes, 2023-10-12:
> @5axes, I believe this is the issue [you've reported in the UltiMaker Community](https://community.ultimaker.com/topic/44294-ultimaker-cura-55-beta-released/?do=findComment&comment=326232).
> 
> Do you agree?

Yes exactly. Temporary workaround : delete the gradualflow Plugin ( un the folder share/cura/plugins)

jellespijker, 2023-10-12:
Instead of deleting thebplugin you could also replace this file https://github.com/Ultimaker/CuraEngine_plugin_gradual_flow/blob/main/CuraEngineGradualFlow/GradualFlowPlugin.py 

It contains a fix such that Cura doesn't crash

5axes, 2023-10-13:
Ok issue solved with the [GradualFlowPlugin.py](https://github.com/Ultimaker/CuraEngine_plugin_gradual_flow/blob/main/CuraEngineGradualFlow/GradualFlowPlugin.py) modification.

MariMakes, 2023-11-16:
To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. I'll be closing this issue.
Rhanks again, and please let us know if you run into any other issues 💪

closed 2023-11-16 by MariMakes