Ultimaker/Cura #16944: Distortion of wall trajectories

Dogm, 2023-10-11:
### Cura Version

all 5.x.x, include  5.5.0 beta 1

### Operating System

Windows 10/11

### Printer

Ultimaker S5/ All

### Reproduction steps

[RIGHT_part_middle_ultimaker_s5_rotated.zip](https://github.com/Ultimaker/Cura/files/12869962/RIGHT_part_middle_ultimaker_s5_rotated.zip)

1. Ultimaker S5 printer profile
2. Layer Profile Fast (0.2 mm)
3. load model
4. slice

see walls and defects on external side of model

### Actual results

[image]

Defects appear at the same height, regardless of repeated slicing

Same defects on Prusa slicer

### Expected results

[image]

Have no defects on Simplify 3D

### Add your .zip and screenshots here ⬇️

Model:
[RIGHT_part_middle_ultimaker_s5_rotated.zip](https://github.com/Ultimaker/Cura/files/12870011/RIGHT_part_middle_ultimaker_s5_rotated.zip)

Cura 5.5.0 beta 1 screen with bug
[image]

Prusa with bug
[image]

Simplify 3D without bug
[image]

Dogm, 2023-10-12:
Another example of this problem on another model

[Segment_part_middle_ultimaker_s5_rotated.zip](https://github.com/Ultimaker/Cura/files/12879971/Segment_part_middle_ultimaker_s5_rotated.zip)

Same scenario for reproduce
[image]

MariMakes, 2023-10-12:
Hi @Dogm,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

I took a look at it in Cura 5.5 and I can reproduce your issue. 
It reminds me of  
[image]

but that one should have been resolved. here: https://github.com/Ultimaker/CuraEngine/pull/1927
I'll bring it up to the team to see what they can do to improve it.

Fingers crossed 🤞

MariMakes, 2023-10-16:
Hi @Dogm,

We discussed your issue with the team.
This seems to be introduced by the Arachne Engine in Cura 5.0.
​​​​​​There might be settings that you can change to improve this behavior.
You can read more about them here: 
https://support.ultimaker.com/hc/en-us/articles/4792077687068

This is still something we want to resolve, so we've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-11167

Thanks for the report! 👍

Dogm, 2023-10-17:
Hi @MariMakes 

Thank you for researching this issue.
I tried to change all the settings from that article and did not get the result of getting rid of this problem, I decided to write here.
I hope that you will be able to correct the generation of wall trajectories in the future

Thank you so much for working with problems 👍

Dogm, 2024-07-06:
Hi, @MariMakes

Have same problem with Cura 5.7.0
but this problem does not exist if you use Surface Mode = Surface

Mode = Normal
[image]

Mode = Surface
[image]

Hopefully this will help in solving this problem....