Ultimaker/Cura #16945: Artifacts in Cura 5.4

BigWampa1989, 2023-10-11:
### Cura Version

5.4.0

### Operating System

Windows 10 - Klipper

### Printer

Creality Ender 3 

### Reproduction steps

I print DND minis and I have been using Cura 5.2.1 but decided to try again after blobbing issues with 5.3 (see INSERT). When I printed the miniarture though it formed these strange indentations in the cape of the model. In reviewing the G-Code preview I could see these errors on the Cura preview. I have attached photos from the 5.4 slice, photos from a correct 5.2 slice, the G-Code itself from 5.4 and the profile from 5.4 as well as photos of the model printed on 5.2 and 5.4. In both tests I am using the standard resolution settings. 

I believe this issue may be linked to another ticket #16207 and perhaps an older ticket I created under Cura 5.3 ticket #14811.

### Actual results

Strange Artifacts in the G-Code forming indentations. 

### Expected results

No artifacts. 

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]
[image]
![Cura 5 2 Cape Front](https://github.com/Ulti
[Cura 5.2 GCODE.zip](https://github.com/Ultimaker/Cura/files
[image]
[image]
[image]
[image]
/12870009/Cura.5.2.GCODE.zip)
maker/Cura/assets/124420868/679c2d22-70b7-4956-a471-ca219bec7418)

MariMakes, 2023-10-12:
Hey @BigWampa1989,

Thanks for your report 👍
Do I understand correctly that the artifacts you refer to are shifted lines?

In the recently released Cura 5.5 Beta we introduced new settings that can be tuned to try to combat this a little, especially if you have a klipper printer. The settings I'm referring to are

- Enable Fluid Motion
- Fluid Motion Shift Distance
- Fluid Motions Small Distance
- Fluid Motion Angle

Could you try the new Cura version with these settings and let us know if they help you? 
You can download the Cura version here: https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1

Another thing you might try are Arachne Enigine specific settings since the walls are thin and seem to have wedges. 
You can read more about them here: 
https://support.ultimaker.com/hc/en-us/articles/4792077687068

BigWampa1989, 2023-10-12:
Why does the 5.2 version of Cura seem to handle this fine? Is there some
sort of trade off occuring?

I will try 5.5 and let you know how it goes.

BigWampa1989, 2023-10-18:
5.5 seams to have resolved the issue. I will update with any new
information.

MariMakes, 2023-11-08:
Hey @BigWampa1989,

To me, it doesn't seem like a bug in Ultimaker Cura anymore. Would you still like to keep this issue open or do you agree that we can close it?

BigWampa1989, 2023-11-08:
I actually have an update on this as I realized it is related to a
different setting. I just have not had the chance to post all the details.
In summary though, it appears to be related to the minimum wall line width
setting under the "Wall" tab which may or may not be as intended. When the
minimum wall line width in the Wall settings was set to 0.2 it caused the
artifacts to appear and then they disappeared when it was set to 0.34. The
uniform pattern of the artifacts though seems to be glitch related though,
no?

Best Regards,

*Louis A. Casadia, Esq.*
***@***.***