Ultimaker/Cura #16950: Cura 5.5 beta.1 crashes when hitting Ctrl-O or selecting File>Open Files

oliof, 2023-10-11:
### Cura Version

5.5.0-beta1

### Operating System

POP!_OS Linux 20.04

### Printer

does not matter

### Reproduction steps

1. Open Cura 5.5.0-beta1
2. hit Ctrl-O (or use the Files > Open File(s) menu dialog)

### Actual results

Cura crashes. When starting from the command line, I see the following errors logged:

```
hwa@pop-os:~/AppImages$ ./UltiMaker-Cura-5.5.0-beta.1-linux-X64.AppImage 
PyInstaller/loader/pyimod02_importers.py:352: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Gtk-Message: 22:22:10.480: Failed to load module "xapp-gtk3-module"
Gtk-Message: 22:22:10.808: Failed to load module "canberra-gtk-module"
Gtk-Message: 22:22:10.812: Failed to load module "canberra-gtk-module"
Cyclic dependency detected between "file:///tmp/.mount_UltiMa5ftoXP/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltiMa5ftoXP/share/cura/resources/qml/Actions.qml"
qt.qml.context: file:///tmp/.mount_UltiMa5ftoXP/share/cura/resources/qml/Settings/SettingView.qml:427:16 Parameter "index" is not declared. Injection of parameters into signal handlers is deprecated. Use JavaScript functions with formal parameters instead.
file:///tmp/.mount_UltiMa5ftoXP/share/cura/resources/qml/PrintSetupSelector/Custom/CustomPrintSetup.qml:163:13: QML ProfileWarningReset: Binding loop detected for property "width"
file:///tmp/.mount_UltiMa5ftoXP/share/cura/resources/qml/PrintSetupSelector/Custom/CustomPrintSetup.qml:163:13: QML ProfileWarningReset: Binding loop detected for property "width"
file:///tmp/.mount_UltiMa5ftoXP/share/cura/resources/qml/PrintSetupSelector/Custom/CustomPrintSetup.qml:163:13: QML ProfileWarningReset: Binding loop detected for property "width"

(UltiMaker-Cura:38528): Gtk-WARNING **: 22:22:40.479: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Pop/16x16/status/image-missing.svg: Unable to load image-loading module: libpixbufloader-svg.so: libpixbufloader-svg.so: cannot open shared object file: No such file or directory (gdk-pixbuf-error-quark, 5)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Pop/16x16/status/image-missing.svg: Unable to load image-loading module: libpixbufloader-svg.so: libpixbufloader-svg.so: cannot open shared object file: No such file or directory (gdk-pixbuf-error-quark, 5)
Aborted (core dumped)
```

### Expected results

File Picker dialog should appear.

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/Ultimaker/Cura/files/12874356/cura.log)

oliof, 2023-10-11:
libpixbufloader-svg.so is not in the AppImage

```
hwa@pop-os:~/AppImages$ ./UltiMaker-Cura-5.5.0-beta.1-linux-X64.AppImage --appimage-offset 
193728
hwa@pop-os:~/AppImages$ sudo mount -o loop UltiMaker-Cura-5.5.0-beta.1-linux-X64.AppImage -o offset=193728
mount: UltiMaker-Cura-5.5.0-beta.1-linux-X64.AppImage: can't find in /etc/fstab.
hwa@pop-os:~/AppImages$ sudo mount -o loop UltiMaker-Cura-5.5.0-beta.1-linux-X64.AppImage /tmp/app/ -o offset=193728
hwa@pop-os:~/AppImages$ cd /tmp/app/
hwa@pop-os:/tmp/app$ find . -name libpixbufloader-svg.so
hwa@pop-os:/tmp/app$ 
```

it is available on the host system

```
hwa@pop-os:/tmp/app$ find /usr/lib -name libpixbufloader-svg.so
/usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-svg.so
hwa@pop-os:/tmp/app$ 
```

The Cura 5.4 AppImage works (and does not have libpixbufloader-svg.so included either)

HellAholic, 2023-10-12:
Hey @oliof,

This is a known issue with the new AppImage. It relates to the Icon sets of the operating system. We're currently working on resolving this issue.

In the meantime, to address your issue, I have the following workaround if you're interested:
- Try using the Default Pop icon set [I've tested this one and it worked from fresh install of POP 20.4 (Intel)]
- Switch to an alternate icon set to see if that works [High contrast set for me crashes while the Pop works]
[image]

Let me know if that works for you,
Thanks for the report! 👍

oliof, 2023-10-12:
Trying another icon set completely broke my UI until I removed all the gnome config directories. So I wouldn't say it helped.

LoeschMaximilian, 2023-10-15:
```
PyInstaller/loader/pyimod02_importers.py:352: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///tmp/.mount_curavpaUzG/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_curavpaUzG/share/cura/resources/qml/Actions.qml"
file:///tmp/.mount_curavpaUzG/share/cura/resources/qml/PrintSetupSelector/Custom/CustomPrintSetup.qml:163:13: QML ProfileWarningReset: Binding loop detected for property "width"
file:///tmp/.mount_curavpaUzG/share/cura/resources/qml/PrintSetupSelector/Custom/CustomPrintSetup.qml:163:13: QML ProfileWarningReset: Binding loop detected for property "width"
file:///tmp/.mount_curavpaUzG/share/cura/resources/qml/PrintSetupSelector/Custom/CustomPrintSetup.qml:163:13: QML ProfileWarningReset: Binding loop detected for property "width"
[1]    14297 segmentation fault (core dumped)  cura
```

I see a similar Issue, but without any warnings concerning any Icons or related Libs. It just crashes attempting to open an open-file dialogue. Open Files, via Recently Opened Files, works without any Problems.

HellAholic, 2023-10-15:
@LoeschMaximilian your observations are correct. The reason it causes a crash is not the open file function, so you can even drag and drop a STL or a 3mf file into Cura and it loads. The cause, as far as we have narrowed it down, is the pop up window for Open file or Save project/file as it is looking for specific set of icons in the current active UI of the OS.

Since Cura crashes, it does not log the issue in the log file, that is why you don't see any warnings or Errors in the log file. We've checked the issue on Mint OS 20.1 xfce & Mint OS 20.3 Cinnamon as well as few other distros based on Ubuntu. When the crash happens during the Open file dialog the cause is the missing icon issue.

You can try an alternative icon set, but there might be underlying issues with different icon sets as oliof mentioned.
For Mint OS we tried "Mint X" and it resolved the issue, and on POP! fresh install, the Pop icon set worked but caused issue for oliof.

[image]
[image]

oliof, 2023-10-16:
@HellAholic I am on POP!_OS 22.04 where the gnome settings look slightly different. I tried another way of setting another icon set (I only can pick standard or high contrast now), and this time that worked. But Cura crashes on slicing -- another bug I haven't had time to investigate for a proper report yet.

LoeschMaximilian, 2023-10-17:
I looked more into my system, I do not have the library @oliof mentioned installed on my system, nor any icon-packs, that I can find. I do not run a desktop-environment, but just a simple Display-Manager under X11.

c---, 2023-10-25:
Running Arch Linux I did `ln -s /usr/lib/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-svg.so /usr/lib` and it seems to fix things. libpixbufloader-svg.so should probably be added to the AppImage?

I'm using the Elementary icon set.

HellAholic, 2023-10-25:
Thanks for all your input and feedback regarding this issue. We've addressed the issue in the Stable release of 5.5

https://github.com/Ultimaker/Cura/releases/tag/5.5.0

Please let me know how it turns out ;)

c---, 2023-10-25:
@HellAholic  The 5.5.0 release does seem to work for me so far without needing to do anything special.

zapp42, 2023-10-26:
5.5.0 still crashes on Fedora 38 and 39.

c---, 2023-10-26:
@zapp42 Works for me using the Fedora 38 Live DVD. I wonder if there could be some system difference. 

One thing I was doing when testing was using:
```
$ strace -o cura.log ./UltiMaker-Cura-5.5.0-linux-X64.AppImage
```
Then looking at the log to see what was failing.

gdb might show something too if it's segfaulting but gdb doesn't always work so well with AppImage:
```
$ gdb ./UltiMaker-Cura-5.5.0-linux-X64.AppImage
(No debugging symbols found in ./UltiMaker-Cura-5.5.0-linux-X64.AppImage)                                                                                                                            
(gdb) run
...
...
<segfault>
(gdb) bt
Backtrace...
```

oliof, 2023-10-27:
Works for me now, thanks!

oliof, 2023-10-27:
closing for now.

closed 2023-10-27 by oliof