Ultimaker/Cura #16951: 5.5.0 Beta won't slice model

alavabeer, 2023-10-11:
### Cura Version

5.5.0

### Operating System

Windows 10

### Printer

Artillery Genius

### Reproduction steps

1. Downloaded model - https://www.thingiverse.com/thing:3251366
2. Sliced it in Cura 5.0.0, worked perfectly
3. Opened Cura 5.5.0, failed to slice

### Actual results

Error message - Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker.

### Expected results

Model should have sliced ready for print.

### Add your .zip and screenshots here ⬇️

[AG_Christmas_Village.zip](https://github.com/Ultimaker/Cura/files/12874480/AG_Christmas_Village.zip)
[image]
[image]

GregValiant, 2023-10-12:
Thanks for the report.
That model has shown up here in a previous bug report.
It would not slice for me "as recevied".  I rotated the model 90° CCW and it sliced.
@MariMakes of the Cura team will take a look.

Your line width and layer height are keeping a lot of fine details (and a couple of roofs) from printing.
It would be a shame to get this far and have one of the last things fail.
This really needs to be printed with a 0.3 nozzle and line width.
This is at 0.20 layer height and a 0.35 line width.  That's about as far down as you can go with a 0.4 nozzle.  As you can see the roof of the topmost building won't slice and the spire would fail.
[image]

This is with a 0.3 nozzle and line width.  The print time is up to 35 hours though.
[image]

KarinaLaursen, 2023-11-04:
I just want to add, i have been trying to slice this file also, no luck, i tried the same settings as on the photo and tried to rotate 90 degrees :(

GregValiant, 2023-11-04:
@KarinaLaursen if you install 4.13.1 it will slice this.  The page is [HERE](https://github.com/Ultimaker/Cura/releases?page=2).  Scroll down to 4.13.1 and click on the "Assets" link.
If you want to stick with the Arachne versions of Cura then 5.2.2 will slice this.
You don't have to un-install a version to install another version.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic