Ultimaker/Cura #16969: Upon Click the Support Blockers Appear Directly in front of Camera, not on Object

mam326, 2023-10-12:
### Cura Version

5.5.0-beta.1

### Operating System

windows 10

### Printer

creality ender 3

### Reproduction steps

I used to be able to select support blocker on the menu, then click on my model, and the blocker would appear on the model very close to where I need it.   Now with the new beta version of Cura I click and the block appears directly in my face hundreds of millimiters away from the model.

### Actual results

[image]
[image]

### Expected results

Support blocker directly on the model like it used to, first click, first try.

### Add your .zip and screenshots here ⬇️

see above
[image]

MariMakes, 2023-10-16:
Hey @mam326,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

I don't believe we changed anything related to support meshes in this release. 
I tried to reproduce your issue in Cura 5.5 Beta but I'm having some trouble reproducing the behavior you are describing, which also makes it hard to test if we implement a fix. 😞 

Do you perhaps have some more precise steps to reproduce or a screen recording of you encountering the issue? 🙏

fieldOfView, 2023-10-17:
The key here is the message in the bottom of the window; the scale of the model is very large, and apparently something is going wrong adding the blocker-geometry as a child of the scaled-up model. I don't think the blocker is actually placed very close to the camera, but the blocker is extremely large. Can you confirm that's the case, @mam326 ?

mam326, 2023-10-24:
Interesting thoughts.  

In fact, I'm now wondering if its not an issue with Plasticity STL exports.  However, its not the fact that the blockers are gigantic, i.e., if my model is scaled up by 1,000,000% like it says, and the blocker is being scaled in kind.  The blocker is created in the usual size, 10x10x10 or whatever, its just literally right in your face, imagine the camera sensor plane, or better yet, imagine holding a camera about 10 feet away from your subject, its like the blocker is being created literally on the front optic of your camera lens, a fraction of a milimeter away instead of 10 feet away. Wheras the normal behavior is for the blocker, upon clicking your model, to appear on the model.

With the previous versions of Cura I didn't have the problem, but I did always have this phenomenon of the 1,000,000% scaled issue, however, my models always printed out perfectly so I just chalked it up as a quirk of some kind.  I googled this issue and someone had the problem with his Blender settings not being millimeter by default and there was a scaling issue with blockers.  I just don't think thats my issue exactly, since Plasticity has always been set to millimeters, and this never happened before.

But if no one else is having this problem, I bet it has something to do with Plasticity exports.  *Or something I'm doing wrong.

I ran the older version of Cura a few minutes ago, and added a blocker, and it works fine.  So its only happening with this Beta.

onack, 2023-11-02:
I had this issue right now in the 5.5.0 release version. Restarting the application solved the issue. I was able to place two blockers, slice and when i went back to prepare to add more they appeared in front of the camera. Model is 106x81x102mm 100% size. Hope this helps to locate the issue. I also have an issue with support blockers not blocking a particular support, but thats another thing

RWSWARD-r1, 2023-11-03:
I am having the same issue now.

massis87, 2024-07-12:
I'm having the same issue intermittently since early version 5's, currently using 5.6.0 and it still occasionally happens. Wherever I click, the support blocker will always appear at the location of the camera instead of the location of the click. The only solution is generating it and then moving it by hand onto the desired position. 

It's very hard to consistently reproduce the issue, but it seems related to the export from Fusion360. I have 4 nearly identical models, all derived from a single model. I exported all 4 one by one, after which I perform the same actions: rotate them to a plane and add 4 support blockers. Worked fine on 2 models, had the camera bug on the 3rd one and it worked fine again on the last one.