Ultimaker/Cura #16971: 15mmx15mmx10mm model fails to slice

Thinkersbluff, 2023-10-12:
### Cura Version

5.4.0

### Operating System

Windows11 v 22H2

### Printer

CR6-Klipper

### Name abnormal settings

Other models slice ok with same settings.  Only this model fails to slice.

### Describe model location

No

### Describe your model

Meshtools Check Models says is watertight.  3DBuilder says is ok.  MeshMixer says is ok.

### Add your .zip here ⬇️

[Cura5.4.0_wandcrown_Project.zip](https://github.com/Ultimaker/Cura/files/12887902/Cura5.4.0_wandcrown_Project.zip)

Model slices ok if scaled-up to 120% or more of original size.

GregValiant, 2023-10-14:
Thanks for the report.
The 5.4 bug could often be fooled by rotating the model, moving the model, or changing the Initial Layer Height but a little bit.  See if any of those work for you.  I didn't have any trouble slicing it at 100%.

(There have been some bug reports here that I'm starting to suspect might have something to do with Windows 11.)

Thinkersbluff, 2023-10-16:
No help to move or rotate the model, but changing the 0.3mm layer height to 0.2mm does allow me to slice the print.

Initial layer height of 0.3mm works fine.

I first found this bug while evaluating 5.5.0-beta 1.  Rolled-back to 5.4.0 and discovered the bug impacted me there, too.

Thanks for the tip.
Hope the root cause can be fixed, soon.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic