Ultimaker/Cura #16974: [Cura] - 100% CPU and file not sliced in 5 hours

khpetrov, 2023-10-13:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Any printer

### Reproduction steps

1. Download the STL file, from the attached zip
2. Open it with Cura
3. Load any profile
4. Load any printer
5. Slice it

### Actual results

CPU goes up to 100% and stays at max until slicing is canceled. The file cannot be sliced in more than 5 hours. 
The same file with Prusa is processed in less than 30 seconds.

### Expected results

The file to be sliced in a comparable time to Prusa.

### Add your .zip and screenshots here ⬇️

[drying_pad.zip](https://github.com/Ultimaker/Cura/files/12892302/drying_pad.zip)

GregValiant, 2023-10-13:
Thanks for the report.
That's pretty bad.  I ran a race between 4.13.1, 5.4, 5.5, and Prusaslicer.  Prusaslicer finished, had dinner, and watched a movie before 5.4 finished.  I gave up on 4.13 and 5.5.
A similar model came through here a couple of weeks ago and it had a lot of square shapes on the build plate.  Turning off "Optimize Wall Printing Order" made a big difference on that one.  I tried that with your model.  It didn't really make a dent.

@MariMakes and the Cura team will take a look at this one.

khpetrov, 2023-10-14:
Thank you for the prompt response. I am so sorry to hear that such an issue
comes straight from Cura core, and it exists in pretty much all versions of
Cura. I really like using Cura, but having a major malfunctioning issue is
more than a solid reason for me to reconsider my priorities for using other
slicing software. I would never give up of Cura if it had some chance to
compare with Prusa, but half a minute slicing time for Prusa vs many hours
for Cura or even days...they are not in the same category. I really hope
that Ultimaker will demonstrate enough will to improve their product
performance.

Best Regards,
Koko

MariMakes, 2023-11-16:
Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/17104
We have a ticket on our backlog with the intent to improve this behavior. 💪 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/17104

closed 2023-11-16 by MariMakes