Ultimaker/Cura #16979: Cura 5.5.0beta1 AppImage does not work on Debian

cdiener, 2023-10-13:
### Cura Version

5.5.0beta1

### Operating System

Linux (Debian Sid)

### Printer

Artillery Sidewinder X2

### Reproduction steps

Download the the AppImage from the release page and make it executable. Run it.

### Actual results

Start will fail with the following messages:

```
 ~/Downloads  ./UltiMaker-Cura-5.5.0-beta.1-linux-X64.AppImage 
PyInstaller/loader/pyimod02_importers.py:352: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
libEGL fatal: DRI driver not from this Mesa build ('23.0.4-0ubuntu1~22.04.1' vs '23.2.1-1')
```

No log file will be generated.

### Expected results

Cura should start. Previous modern AppImages work fine.

### Add your .zip and screenshots here ⬇️

No log file has been generated.

hjri, 2023-10-14:
same here on manjaro
```
libEGL fatal: DRI driver not from this Mesa build ('23.0.4-0ubuntu1~22.04.1' vs '23.1.8-manjaro1.1')
```

luismanson, 2023-10-14:
Hello, I have the same issue with OpenSuSE Tumbleweed. 
I found another issue with another build that works for me: https://github.com/Ultimaker/Cura/issues/16932#issuecomment-1757681042

mechano, 2023-10-15:
Same error on new Ubuntu 23.10.

MariMakes, 2023-10-16:
I recognize your Ubuntu version 🤔
Ubuntu does not ship FUSE 2 anymore on 23.04. This is required to run our appimages.
We have reports that Cura 5.4 runs if you install FUSE 2 with sudo apt install libfuse2

Can you let us know if installing this library works for you? 

We are looking into improving our Linux build for 5.5 stable, could you confirm if https://github.com/Ultimaker/Cura/issues/16932#issuecomment-1757681042 works for you, or at least if you believe it's a duplicate?

mechano, 2023-10-16:
> confirm if #16932 (comment)

It works with X.Org, I already have libfuse2 installed. It seems a problem
related to 3D graphics library on Wayland.

Hello1024, 2023-10-18:
I suspect this commit is the cause:

https://github.com/Ultimaker/Cura/commit/d8ef4c381c4832a91c9b0dcaa96ba3eb0b6704d1

It removes libegl from the appimage - and therefore presumably libegl is being picked up from somewhere else, and is the wrong version.

mechano, 2023-10-18:
> d8ef4c3

5.6.0 alpha works. There's something different.

cdiener, 2023-10-21:
> Ubuntu does not ship FUSE 2 anymore on 23.04. This is required to run our appimages.
> We have reports that Cura 5.4 runs if you install FUSE 2 with sudo apt install libfuse2

As others have mentioned it is not related to libfuse2 which is installed on my system as well and other app images work.

> We are looking into improving our Linux build for 5.5 stable, could you confirm if https://github.com/Ultimaker/Cura/issues/16932#issuecomment-1757681042 works for you, or at least if you believe it's a duplicate?

The link in this comment does not work anymore because the PR has been merged. However downloading the AppImage artifact from the latest Actions run does work. It is very sluggish though and takes several minutes to startup (previous versions start up within a few seconds).

jellespijker, 2023-10-26:
We did a couple of fixes for the AppImage, can people try the 5.5 stable AppImage and report back if their issues are fixed?

thx

cdiener, 2023-10-31:
> We did a couple of fixes for the AppImage, can people try the 5.5 stable AppImage and report back if their issues are fixed?
> 
> thx

Yes the stable AppImage works for me, thanks. Closing this here because the original issue was fixed.

closed 2023-10-31 by cdiener