Ultimaker/Cura #16980: Sliced fine at 100% and 50% but then failed at 75% scale

cellocgw, 2023-10-13:
### Cura Version

5.4

### Operating System

macOS 13.5.2 ventura

### Printer

anycubic vyper

### Describe your model

yes, at different scales

### Add your .zip here ⬇️

[Upper SkullTest4.3mf.zip](https://github.com/Ultimaker/Cura/files/12899325/Upper.SkullTest4.3mf.zip)

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic

cellocgw, 2024-04-03:
5.7 beta slices this and others at all scales tested.  Thanks for the fix