Ultimaker/Cura #16981: File from Thingiverse failed slicing in Cura 5.4

DaviddKirk, 2023-10-14:
### Cura Version

5.4

### Operating System

Windows10

### Printer

LK5 Pro

### Name abnormal settings

None that I know of

### Describe model location

No.

### Describe your model

Yes, but only after several failed attempts

### Add your .zip here ⬇️

[LLK5PRO_Bottom.zip](https://github.com/Ultimaker/Cura/files/12900625/LLK5PRO_Bottom.zip)

GregValiant, 2023-10-14:
Thanks for the report.
The model has some errors.  Open edges and inverted faces will confuse a slicer.
This is from formware.co/OnLine STLrepair
-> Analysed your file:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> 12 Non-manifold edges (?)
--> 21 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 1491ms ------
-> Vertex count changed from 1848 to 1991 (+143)
-> Triangle count changed from 3756 to 4042 (+286)

After repairing the model I still had to rotate it to get it to slice (that's a bug), but it sliced.
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic