Ultimaker/Cura #16986: slicing failed on Front_LowerAA

Lukens4242, 2023-10-14:
### Cura Version

5.5.0-beta1

### Operating System

Linux

### Printer

Voron2.4r2

### Name abnormal settings

layer height of .1 or .15

### Describe model location

centered, moving and rotating doesn't help

### Describe your model

It slices fine at .2mm layer height but not less.

### Add your .zip here ⬇️

[Front_LowerAA.zip](https://github.com/Ultimaker/Cura/files/12907965/Front_LowerAA.zip)

Lukens4242, 2023-10-14:
This will slice properly in 5.3.1, but not in 5.4 or 5.5.0-beta1

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic