Ultimaker/Cura #16987: Unable to open File, The File could be corrupt, inaccessible or it did not contain any models

ZigeunerBro, 2023-10-14:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Kobra Plus / Ender 3 S1 Pro

### Reproduction steps

I tried to print a STL file at the following site: https://wiki.openastrotech.com/OpenAstroTracker/Addons/AutoPA

The files are available on github: https://github.com/OpenAstroTech/OpenAstroTracker-Addons/tree/master/AutoPA/STL/V2.0

Every time I try to open one of these files, nothing happened. only an error msg occur an says "Unable to open File, The File could be corrupt, inaccessible or it did not contain any models"

I tried different versions of cura and in version 4.4 (i believe), i could open one of the files. After that I was able to open this STL in 5.4.0 without issues.

### Actual results

Error Msg:
"Unable to open File, The File could be corrupt, inaccessible or it did not contain any models"

### Expected results

just open the file

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/Ultimaker/Cura/files/12908174/cura.log)

GregValiant, 2023-10-15:
Thanks for the report.
I downloaded the "13_rear_AltAz_mount_6801.stl" file and didn't have any trouble.  I have had that error message come up when I try to load a 3mf Project File and the model is missing.  I've never seen it with an STL file.
Is it possible that your anti-virus or firewall is having an effect on the download?
[image]

ZigeunerBro, 2023-10-15:
Thx for your help.

[image]
[image]

I found the issue.

I downloaded the files via right click/save as... in the file overview section. And that seems to be the problem.

If i click on the file and use the download button, everything is fine.

GregValiant, 2023-10-16:
I'll go ahead and close this as solved.

closed 2023-10-16 by GregValiant