Ultimaker/Cura #16991: Cura often stops gcode near the end of a print, wasting a ton of filament.

Floatonokay, 2023-10-15:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Anycubic Kobra

### Reproduction steps

1. Slicing a model from thingiverse.com
2. Saving to sd card for printing
3. The gcode often stops suddenly right before finishing, wasting hours of printing and filament.

I have no idea why and it doesn't happen every single time but the slice does not complete even though it shows that it is complete in Cura. I honestly thought I had even checked this gcode before printing but maybe I just missed it--the top part of the statue's hair is missing. 

### Actual results

The print stops early and is completely ruined.

### Expected results

The print should just work. 

### Add your .zip and screenshots here ⬇️

[image]

MariMakes, 2023-10-16:
Hi @Floatonokay,

Welcome to the UltiMaker Cura Github 🚀 
Aww, that looks super annoying, especially since it's so close to being done. 😞 

This looks like a USB running out of space, or the USB was removed before Cura finished saving. 
In the Cura 5.5 beta we introduced a warning error message to reduce the cases. 
You can download the Cura version with the fix here: https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
Would you still like to keep this issue open or do you agree that we can close it?

Floatonokay, 2023-10-16:
Hi thanks for the answer!

Your solution totally makes sense but my SD card has plenty of space every
time this happens. Maybe issue with the memory itself? I’ll try to
replicate with a different SD card.

Feel free to close if you feel confident this is a memory issue. Thanks for
your help!

MariMakes, 2023-10-24:
To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
'll be closing this issue. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2023-10-24 by MariMakes