Ultimaker/Cura #16992: Cura 5.5.0-beta.1 appimage : crash when importing

didli, 2023-10-15:
### Cura Version

5.5.0-beta.1

### Operating System

Fedora  38

### Printer

Creality Ender 3 neo v2

### Reproduction steps

Note : this is just a report. This is a beta release and I don't expect it to fully works anyway.

- Trying to import something (from open file button or menu) will result in an immediate crash and app force close.

### Actual results

- Trying to import hangs a bit at first, then cura 5.5-beta.1 crashes.
- No issue in cura 5.4 stable release

### Expected results

- Import should work as the previous stable release

### Add your .zip and screenshots here ⬇️

This is the log when running the appimage from a terminal : 
```log
./ultimaker-cura-5.5.0-beta.1-linux-x64.appimage
/usr/bin/AppImageLauncher: /usr/lib64/libcurl.so.4: no version information available (required by /usr/bin/../lib/x86_64-linux-gnu/appimagelauncher/libappimageupdate.so)
PyInstaller/loader/pyimod02_importers.py:352: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Gtk-Message: 14:03:25.276: Failed to load module "xapp-gtk3-module"
Cyclic dependency detected between "file:///tmp/.mount_ultimaCbMK2M/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_ultimaCbMK2M/share/cura/resources/qml/Actions.qml"

(UltiMaker-Cura:46868): Gtk-WARNING **: 14:03:46.845: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.

-// crash happens here //-
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Papirus-Dark/16x16/actions/image-missing.svg: Unable to load image-loading module: libpixbufloader-svg.so: libpixbufloader-svg.so: cannot open shared object file: No such file or directory (gdk-pixbuf-error-quark, 5)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/Papirus-Dark/16x16/actions/image-missing.svg: Unable to load image-loading module: libpixbufloader-svg.so: libpixbufloader-svg.so: cannot open shared object file: No such file or directory (gdk-pixbuf-error-quark, 5)
```

jellespijker, 2023-10-15:
devs and @MariMakes see CURA-11132

MariMakes, 2023-10-16:
Hi @didli,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/16932
We have a ticket on our backlog with the intent to improve this behavior, and if you check the ticket, you can even find a build that potentially resolves this issue. 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/16932

Thanks for your report. I believe that together we can make Cura better 😊

closed 2023-10-16 by MariMakes