Ultimaker/Cura #16993: Tool change bug when not using prime tower, since many years

svarteld, 2023-10-15:
### Cura Version

5.5.0 beta

### Operating System

Mac OS 12.6.1

### Printer

Rasie 3D Pro2 Plus, FFCP, and any other dual head printer

### Reproduction steps

Sine many years, including this latest beta, when issuing a tool change in Cura and not using a prime tower, the nozzle still moves to the exact coordinates where the prime tower would have been; the coordinates specified in Cura under Dual Extrusion, Prime Tower X Position, Prime Tower Y Position. These settings are only visible with Enable Prime Tower On, but still moves the nozzle to those coordinates when the Enable Prime Tower is deselected.

See the two screen dumps. I've verified this many times in real life and in gcode. It's driving me nuts!

If using Avoid Printed Parts When Traveling, it's not even script-removable, since many moves will have to be deleted, not just the G0 to the hidden prime tower.

Many people previously thought this was a home-between-tool-change bug, but that's because they've got their non-existent prime tower at origo.

PLEASE, remove this bug now for the 5.5 real release; it should be really easy.

Thanks!

### Actual results

See dumps

### Expected results

See text above

### Add your .zip and screenshots here ⬇️

[image]
[image]

MariMakes, 2023-10-16:
Hey @svarteld,

Welcome to the UltiMaker Cura Github 🚀 

What you describe seems to be something very specific to your type of printer, which is not always something that we can work on in the team, but something that can be resolved by the community. Since Cura is open source, you or anyone else is more than welcome to work on adding a pull request yourself. If you are stuck in any way, please reach out! I'll be happy to help out. 

When I look at your description, it reminds me of this setting that's already present in Cura:
[image]

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

svarteld, 2023-10-19:
Hello MariMakes,

Thanks for greets and for clarifying, appreciated!

Before we spend time on files; is Cura starting every layer at a fixed position for the nozzle, more precisely at Layer Start mentioned above? If that's so, it seems very counterproductive, since it means redundant moves every layer. I realise this wouldmightmake it easier for you to make CPU parallell processing of the slicing easier to program, but it's at the cost of slicing quality.

Thanks /Peter

wawanbreton, 2024-03-22:
Hi @svarteld, and sorry for the answering delay :smiley: 

To answer your question, the starting position of each layer depends on the end position of the previous layer. You are not supposed to see these travels moves. I must admit I don't exactly know what the "Layer Start" position is for, but it looks more like a fallback position that is optimized when possible.

Anyway, the priming behavior has been changed recently, could you try reproducing your issue with the latest 5.7 beta ? And if the issue is still present, please send a 3MF file so that we can easily reproduce it, that really helps !