Ultimaker/Cura #17002: 5.5 beta 1 - Materials section in settings does not open and causes visual glitches in preferences

Slashee-the-Cow, 2023-10-16:
### Cura Version

5.5 Beta 1

### Operating System

Windows 10 x64 22H2

### Printer

Ender-3 V3 SE, but printer independent

### Reproduction steps

In Cura, go to Preferences > Configure Cura
Select the "Materials" section
Materials section does not open
Visual glitches will start appearing if you change settings categories back and forth, sometimes the category you had selected immediately before Materials will appear without glitches, but I have been unable to reliably replicate this behaviour.
If you close preferences then open then again, the glitches remain, it is only reset after restarting Cura
Tried in 5.4 and worked perfectly.

I tried to replicate it on another computer running a clean install of Cura (Surface Pro 2, Windows 10 x64) but it worked as expected, although the Materials section did take a few seconds to load. There are obviously many different factors at play, but I'd like to note that while I've added custom machine definitions and post-processing scripts on my main computer, I've never touched the material profiles, either in the settings or manually in the files.

### Actual results

In preferences, select Materials section, preferences remain on previous screen:
[image]
Go to a different section, in this example Profiles, and I apologise for my highly-Australian proliferation of swearing in the profile titles:
[image]
Notice that the title is glitched (it appears to have placed "Profiles" on top of "General") and the phantom section of button in the top right - it's the same size as the "Defaults" button in the general settings, and it responds to hovers and clicks (results not immediately visible due to it having reset the General preferences to the defaults). The "Import" button works as expected.

Go to another section:
[image]
It's clear that "Settings Visibility" has been written on top of the existing category title. There's no phantom button parts because the "Defaults" button for this section is the same size as the General section.

Go to yet another section:
[image]
Again, no phantom button because the "Add New" button is actually slightly bigger than the "Defaults" button. And the category title has been truncated I assume because the label widget is not as long.

Go back to General category -  title is still messed up but no phantom button from the "Add New" from the printer section.
[image]

### I restarted Cura to document different results.
Open preferences, choose another category (in this case Printers) before Materials, and then select Materials. Materials is selected in list, but Printers settings are still visible.
[image]

Swap to General - title is overwritten and there is part of the button for "Add New" behind the results, and responds to both hovers and clicks (brings up the new printer wizard, mouse cursor is hidden in screenshot but you can see the button has the hover colour background)
[image]

### Expected results

It should behave the same as it does in 5.4 - Materials settings open fine and no visual glitches:
[image]

### Add your .zip and screenshots here ⬇️

As it didn't happen on a clean install, here is the %APPDATA/cura/5.5/ folder from my main computer - I had to remove a couple of the larger plugins to make it small enough to fit in the file size limit but that change had no effect on the visual glitches.
[slashee appdata 5.5.zip](https://github.com/Ultimaker/Cura/files/12916997/slashee.appdata.5.5.zip)

MariMakes, 2023-10-16:
Hey @Slashee-the-Cow,

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/13688
We have a ticket on our backlog with the intent to improve this behavior.
If you check the linked ticket there is a possible workaround there, that might work for you as well. 🤞 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/13688

Slashee-the-Cow, 2023-10-16:
Thanks for that - I did search but apparently my search skills were not sufficient to find that.

Slashee-the-Cow, 2023-10-16:
> Hey @Slashee-the-Cow,
>
> This seems to be a duplicate of #13688 We have a ticket on our backlog with the intent to improve this behavior. If you check the linked ticket there is a possible workaround [...]

This is **not at all** like that other bug. That one just concerns substandard text rendering apparently caused by Windows' subpixel antialiasing of text. That bug has **absolutely nothing** to do with overlaid text and buttons in the settings window and not being able to access the materials settings **at all**.

GregValiant, 2023-10-16:
Hi @MariMakes 
I have this same problem (see my post on the [Forum](https://community.ultimaker.com/topic/44294-ultimaker-cura-55-beta-released/)
The "Materials" dialog appears to be broken (Win10Pro).  When I click on it or select "Manage Materials" from elsewhere the dialog does not open.
I think this report should be re-opened.  Yes, the fonts are wrong in the "Preferences" dialog, but the "Materials" dialog would seem to be a completely separate issue.
Here you can see the bad text on the Preferences dialog, but also note that the "Material" button has been clicked but the General preferences remains active.  The "double default" buttons are also an issue.
[image]

h3rm, 2023-10-17:
same problem here with flsun v400 and wqhd-resolution (2560px)

GregValiant, 2023-10-19:
@MariMakes I'm going to go ahead and re-open this.  I'm not able to access the Materials at all.

GregValiant, 2023-10-20:
@Slashee-the-Cow this appears to be related to a change in Cura that seems to have affected the Materials Plugin.  I'm not sure how this will be resolved but for now un-installing the Materials plugin seems to fix the dialog box problem.
There is more at [FieldOfViews GitHub page.](https://github.com/fieldOfView/Cura-MaterialSettingsPlugin/issues/29)

fieldOfView, 2023-10-20:
It will be resolved by an update of the Material Settings plugin, which I am working on now.