Ultimaker/Cura #17007: Crash on startup

FirewallDaProtogen, 2023-10-16:
### Cura Version

4.13.1

### Operating System

Windows 7 pro (64 bit) Virtual Machine

### Printer

n/a

### Reproduction steps

all i did was install version 4.13.1 onto my windows 7 virtual machine, and once i selected "finish" with "run ultimaker cura" checkboxed

### Actual results

i get a "cx_freeze python error in main script" window opening, plus, no error log is generated.

### Expected results

the program starts

### Add your .zip and screenshots here ⬇️

[image]
[image]

MariMakes, 2023-10-17:
Hey @FirewallDaProtogen,

Welcome to the UltiMaker Cura Github 🚀 
Cura 4.13 is a release that is from January 2020.

We've introduced a lot of changes since then.
You can download the latest version of Cura here: https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1

If you are stuck on this version on purpose, can you help me understand why you can't upgrade?
We are not able to backport any fixes into Cura 4.13, would you still like to keep this issue open or do you agree that we can close it?

FirewallDaProtogen, 2023-10-17:
i was on that version because i was intending to use it alongside other old software i had on a windows 7 machine, upon further investigation the issue is simply that i was using virtualbox, which doesnt support above opengl 1.1

i ended up just installing the latent version on my main machine. thanks for the reply though!

closed 2023-10-17 by FirewallDaProtogen