Ultimaker/Cura #17009: 5.5.0-Beta1

qwerty8224, 2023-10-17:
### Cura Version

5.5.0-Beta1

### Operating System

Windows 11

### Printer

Gustom FFF

### Reproduction steps

We have a problem again 

### Actual results

1

### Expected results

1

### Add your .zip and screenshots here ⬇️

[CFFFP_Spr wheel 1.zip](https://github.com/Ultimaker/Cura/files/12925434/CFFFP_Spr.wheel.1.zip)
[image]

MariMakes, 2023-10-17:
Hey @qwerty8224,

Thanks for your report 👍
We introduced Fluid Motion Settings in Cura 5.5 to help combat these microextrusions. 

You can find them under Mesh Fixes:
[image]

Could you let us know if tuning these settings helps to resolve your issue?

qwerty8224, 2023-10-17:
oops it's my fault (((enabled the settings and the artifacts are gone. Thank you)
problem solved.