Ultimaker/Cura #17010: [5.5.0-b1] Initial layer speed not honored

stestagg, 2023-10-17:
### Cura Version

5.5.0-b1

### Operating System

Mac Sonoma 14.1

### Printer

Creality Ender-3 S1 (pro, using klipper v0.11.0-299-gb1f597c5)

### Reproduction steps

1. Slice a model
2. Preview the first layer in Speed view

### Actual results

3. Note that the speed of outer walls (on layer 1) matches the outer wall speed, not the initial layer print speed:
I.e. not 15mm/s but 150mm/s

[image]

### Expected results

3. Lines on the initial layer all printed at initial layer print speed

### Add your .zip and screenshots here ⬇️

Stl can be provided if useful

[image]

stestagg, 2023-10-17:
Just a note, the produced gcode also has these speeds.  

I noticed the issue because my printer was going mad on the first layer, and nothing was sticking.

stestagg, 2023-10-17:
Just retried on 5.4.0 and it works as expected
[image]

GregValiant, 2023-10-17:
Thanks for the report.
I can't duplicate this with my Ender 3 Pro.  There may be a setting in your definition file that is affecting this.
Please provide a project file so we can do a comparison.
This is in 5.5beta with the settings as I see in your screenshot.  Everything is at "Initial Layer Speed".
[image]

stestagg, 2023-10-17:
That's weird.  The attached project file uses a smaller model, but shows the same issue for me
[image]
[wall-mini.3mf.zip](https://github.com/Ultimaker/Cura/files/12928558/wall-mini.3mf.zip)

GregValiant, 2023-10-17:
Thanks for the file.
My "Slow PLA" profile shows exactly as you show - the Initial Layer speed seems to be ignored.
My "Fast PLA" profile shows everything as expected with the Initial Layer speed enforced.
I think what is happening is that the original "Quality" profile that the custom profiles were created from is still having an effect even though settings are overridden.
If I set Cura to "Dynamic" quality with 2 Slower Initial Layers at 10mm/sec I get this:
[image]
With "Super" quality with 2 Slower Initial Layers at 10mm/sec I get this:
[image]

I haven't been able to find exactly what is causing this.  The Cura team will take a look.
It does appear to be related to the Quality profiles that were provided by Creality as part of the Ender definition files.  That would seem to make it a Creality problem rather than a Cura problem.

stestagg, 2023-10-17:
Thanks for taking the time to look!

For that it's worth, I've just narrowed down the trigger for me: changing the 'Slicing Tolerance' setting back to 'Middle' fixes the problem, but both 'Inclusive' and 'Exclusive' modes cause the fast initial layer problem.

MariMakes, 2023-10-17:
Hi @stestagg,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

This was quite a puzzling issue, but I believe I've figured it out. 
The model you shared seems to be faulty. 
When I loaded it into Microsoft 3D builder, I got prompted a repair message.
In the screenshot, on the right you can see the fixed model, on the left the model you shared.
You can see that the initial layer is green indicating having it's own initial layer speed. 
[image]

This is the repaired model: [CE3S1_wall-mini.zip](https://github.com/Ultimaker/Cura/files/12929344/CE3S1_wall-mini.zip)

To me, it doesn't seem like a bug in Ultimaker Cura, since the error is caused by your model. 
I'll be closing this issue. I hope you understand.

closed 2023-10-17 by MariMakes

stestagg, 2023-10-17:
Hi, Thanks @MariMakes and @GregValiant, and sorry, I've trained myself to ignore the watertight error message, as it's so common now! but it was telling me there was a problem.

I've checked and it's working for me once the mesh is fixed.

Rai80, 2023-11-01:
I've got the exact same issue. I also start noticing this when after the upgrade to 5.5.0. The initial layer wall speed is too high.
I did do the model fix with 3D builder, but this did not fix the problem. In Cura 5.4.0 i dont have this issue.

[image]

Deplerr, 2023-11-18:
I can also confirm on my own end that this is only occurring in 5.5.0 and later releases, I've been pulling my hair out trying to solve it and going back to 5.4.0 seems to have fixed the problem.

5.5.0:

[image]

5.4.0:

[image]

I've tried tweaking every setting to fix this but at the time of writing the only solution seems to be to revert to 5.4.0.

lxlolxl, 2023-11-21:
I got the same problem cura 5.5.0 with my voron2 custom, but with the same model, slice on Flsun Super Racer won't see that problem.

GregValiant, 2023-11-21:
@MariMakes  I would like to re-open this as it is an ongoing problem.  It isn't just an "Initial Layer Speed" problem so maybe changing the report title would be in order.

This is the model submitted by @stestagg on 10-17.  I ran this model through MS 3D Builder and it has no errors.
I have the Print Speed set to 75 and the Flow Equalization is 100% and the line width at 0.65 (0.6 nozzle).  I am achieving the same results in 5.5.0, and 5.6beta2.

You can see that the speed of the mid-wall extrusion is 259mm/sec.
[image]

If I change the Flow Equalization to zero% the problem goes away.
[image]

This behavior appears to involved the wall thickness and the Line Width that will fit between the outer walls. 

This is the same setup in 5.4.0.  You can see that the Flow Equalization at 100% results in a reasonable speed increase to 84mm/sec.
[image]

This is in 5.2.2.  The speed has been adjusted **_down_** to 57.93 rather than **_up_** to 259.  A 200mm/sec speed difference would seem to indicate that _sumthin' ain't right._
[image]

lxlolxl, 2023-11-21:
@GregValiant 
I just found that "Initial Layer Speed" indeed have some problem to solve. 
But it's about the machine's .def.json.

For example, voron2, fine the file :
"UltiMaker Cura 5.5.0\share\cura\resources\definitions\voron2_base.def.json"
Open and delete the line:
"roofing_layer_count": { "value": 1 },
"speed_topbottom": { "value": "math.ceil(speed_print * 0.33)" },

So that I can be sure Initial Layer Speed will work.

ole1986, 2023-12-08:
Does the change  on file `voron2_base.def.json` only applies to the reported "Creality Ender-3 S1" printer model?

EmJay276, 2023-12-08:
> Does the change  on file `voron2_base.def.json` only applies to the reported "Creality Ender-3 S1" printer model?

It's only for Voron based profiles. Have a look at the corresponding Ender file. Sorry I don't know how it is called and can't look it up for you right now.

GregValiant, 2023-12-08:
@ole1986 it does not.  The "creality_base.def.json) file says:
"speed_topbottom": { "value": "speed_print" }
So there is no formula involved like with the Voron.

ole1986, 2023-12-08:
I took a quick look into my related file using the FLsun Q5/QQ and could not quickly figure it out.. so that is why I switched back

GregValiant, 2023-12-09:
5.2.2 would be my choice.

Flight, 2024-02-10:
Getting the same on Sovol SV06, so it not only about Vorons. Flow equalisation and print thin walls options doesn't help.