Ultimaker/Cura #17011: Undertale - Фигурки Темми и надоедливых собак

Sinner207, 2023-10-17:
### Cura Version

5.5.0 beta

### Operating System

Win10

### Printer

anycubic mega s

### Describe your model

Tried it didn't help

### Add your .zip here ⬇️

https://www.thingiverse.com/thing:1252902/files

MariMakes, 2023-10-17:
Hi @Sinner207,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

Have you tried reverting to the default settings?
Have you tried slicing with Support disabled? 
Have you tried moving the model away from the center of the buildplate and rotating by 90 degrees?
Can you check if your mesh is faulty with the Meshtools Plugin? You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

Sinner207, 2023-10-17:
I don’t quite understand how the plugin works, you can’t cut the model with it, but you can cut it if you put it in a different way, but then the printing will not be of very high quality.

I don’t think that the default settings will help, since other models are printed without problems, and changing for the sake of one model makes it easier not to print it.
[image]
[image]
[image]
[image]
[annoyingdog.zip](https://github.com/Ultimaker/Cura/files/12930557/annoyingdog.zip)

Sinner207, 2023-10-18:
Since I didn’t have more than one editor, and fusion 360 doesn’t work in Russia now, I had to install another =) resave this STL and everything worked.

I'm sorry for the trouble.

closed 2023-10-18 by Sinner207