Ultimaker/Cura #17012: Unable to unlink Enable Support Interface

jyoodevil, 2023-10-17:
### Cura Version

5.5.0

### Operating System

Windows 11

### Printer

Custom

### Reproduction steps

Couldn't reproduce it. I don't know what caused it. But the settings are hidden even though they shouldn't be and I can't make them reappear.

### Actual results

Support Roof was enabled. After saving the project file and reopening the file, at some point, the checkbox for enabling the Support Roof and the Support Floor disappeared, and the Enable Support Interface box is grayed out. Now I need to add a Support Floor but the box isn't there and under the Cura settings it says that "This setting has been hidden by the values of Scene Has Support Meshes, Enable Support Floor, Generate Support. Change the values of those settings to make this setting visible." But I'm unable to change those settings or I am changing them wrong.  

### Expected results

I expect for the boxes to not disappear. And if they do, I should be able to toggle the Enable Support Interface to bring them back.

### Add your .zip and screenshots here ⬇️

[Profile bug.zip](https://github.com/Ultimaker/Cura/files/12929529/Profile.bug.zip)

MariMakes, 2023-10-17:
Hi @jyoodevil,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

Do you have a different support pattern than you would usually have?
Do you have tree support enabled? 

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

jyoodevil, 2023-10-19:
I'm not using tree supports and the support pattern isn't too different. That's why I just noticed that some of my print profiles now have the roof and floor check boxes missing and some don't. I deleted my profile and remade it from scratch with all the same settings and now the check boxes are there. Here is a copy of a project with the bugged profile before I remade it.
[am-sld_rail TEST.zip](https://github.com/Ultimaker/Cura/files/13046552/am-sld_rail.TEST.zip)

Thanks