Ultimaker/Cura #17021: The Blob

Torimax, 2023-10-18:
### Cura Version

5.4.0

### Operating System

windows 11

### Printer

Ender 3 Max Neo

### Describe your model

Never been able to do a filament color change without blob issue

### Add your .zip here ⬇️

Hello,
  I have 2 printers, a Ender 3 Pro and a Ender 3 Max Neo. When I try to use the filament change extension, I get a blob of filament on my project. When I use the Pro, it does it before leaving the print, it stops in the corner and purges a blob onto the project. With the Max Neo, It expels filament after returning to the print for the second color. It literally does it every time. I'm sure there is some kind of setting or code I can modify but I couldn't find an answer in the forums. Is this a bug or am I just doing something wrong? Thank you!

GregValiant, 2023-10-18:
Thanks for the report.
If you are using the "Filament Change" post processor then Cura inserts a single command into the gcode.  That M600 line passes the settings to the firmware of the printer and the printer runs a script based on the parameters.  So different firmware can interpret the M600 differently.  All Cura does is put the line together based on the settings and then paste the line into the correct place in the Gcode.  As long as the line is being inserted into the correct place in the gcode then it is unlikely that the blob is a Cura bug.
;BEGIN FilamentChange plugin
M600 E30.00 U300.00 X0.00 Y0.00
;END FilamentChange plugin

M600 is enabled in most firmware, but maybe not exactly the same way.  That would be why you see different effects on the different printers.  You may be able to make a manual adjustment in the M600 line to fix what is happening on the blobby printer.  You will need to experiment.

Pause at Height is different in that Cura puts together a script and inserts that into the Gcode.  It is dependent on the "pause command" being enabled in the firmware (usually M0 or M25 for Creality).  Creality has adjusted their firmware on different models and never tells anyone what they've done so you would need to experiment to find what works for each of your printers.  My printer is an 8bit Ender 3 Pro and Pause at Height works well as M0 is enabled in my firmware.  That isn't always the case on machines with the 4.2.x mainboards.

I'll remove the bug label from this for now (I can put it back on if it turns out this is a bug).  I'll leave it open so you can respond.  You can do some reading about M600 on the [MarlinFW.org](https://marlinfw.org/meta/gcode/) site.

GregValiant, 2026-09-10:
I'll go ahead and close this.

closed 2026-09-10 by GregValiant